zigpy/zigpy-cc #2: [REQUEST] Z-Stack 3 (3.x) firmware support

Gamester17, 2020-01-07:
Can you clarify and confirm if Z-Stack 3 (3.x) firmware will be supported or not by zigpy-cc as it is?

Don't own a CC2531 USB stick yet but would just like to clarify and confirm Z-Stack 3.x support ;)

No pressure or urgency but I am also indirectly asking to know if  CC2538 and CC2652R can be supported in the future too as the Z-Stack 3 (3.x) firmware can also be used on those chips as well.

Gamester17, 2020-01-13:
FYI; ioBroker is another JavaScript based project which Zigbee via same custom Z-Stack by Koenkk 

https://github.com/ioBroker/ioBroker.zigbee

ioBroker claim to not only support CC2531/CC2530 but also CC26x2R and CC2538 (+ CC2592 PA)

https://github.com/ioBroker/ioBroker.zigbee/blob/master/README.md

ioBroker Zigbee module is based on zigbee-herdsman library and their new converters database

puddly, 2020-01-24:
It works fine after [some very minor tweaks](https://github.com/sanyatuning/zigpy-cc/compare/master...puddly:master) with my CC26X2R1. I'm running ~20 devices from my network on it so far and will make a pull request with the changes once I track down why groups of ~four transmissions aren't all succeeding, even with retries.

sanyatuning, 2020-01-25:
Thanks @puddly, you helped me to find a major bug with [clusterid](https://github.com/sanyatuning/zigpy-cc/commit/d8dd2b9b84f529c0c229e04956f231a34f0e6d96#diff-87cd7243ec1834cccd62ee0bc40d22d9R102)

puddly, 2020-01-27:
@sanyatuning I'm just getting familiar with TI's Z-Stack so thanks for letting me know what that code actually does, it makes sense now.

Gamester17, 2020-02-06:
@sanyatuning Just wondering if you would review Pull Request https://github.com/sanyatuning/zigpy-cc/pull/8 by @puddly

sanyatuning, 2020-05-03:
working with zStack 3x0 (tested with zzh adapter)

closed 2020-05-03 by sanyatuning