zigpy/zigpy-cc #4: [REQUEST] Support for CC2652R chips with Z-Stack 3 firmware

Gamester17, 2020-01-07:
Can zigpy-cc support CC2652R chips by Texas Instruments if use Z-Stack 3 based firmware?

CC26x2R chips are much faster and contain much more memory than the CC2531

Texas Instruments development board can be bought http://www.ti.com/tool/launchxl-cc26x2r1

CC2652R is supported Z-Stack firmware by Koenkk https://github.com/Koenkk/Z-Stack-firmware

For reference please see: https://github.com/Koenkk/zigbee2mqtt/issues/1429

Please also see dependency request for Z-Stack 3 (3.x) firmware support https://github.com/sanyatuning/zigpy-cc/issues/2

Gamester17, 2020-01-13:
FYI; ioBroker is another JaveScript based project which Zigbee via same custom Z-Stack by Koenkk 

https://github.com/ioBroker/ioBroker.zigbee

ioBroker claim to not only support CC2531/CC2530 but also CC26x2R and CC2538 (+ CC2592 PA)

https://github.com/ioBroker/ioBroker.zigbee/blob/master/README.md

Gamester17, 2020-02-02:
FYI, read that "zig-a-zig-ah" ("zzh" for short) USB-stick form-factor dev board based on CC2652R / CC2652P has been announced as an open-source hardware design by Electrolama (  @OmerK and friends ) under open source Solderpad Hardware Licence Version 2.0

https://github.com/electrolama/zig-a-zig-ah/

https://github.com/electrolama/docs/blob/master/docs/projects/zig-a-zig-ah.md

"Assembled versions of zig-a-zig-ah! will be made available on Tindie when it is ready"

https://electrolama.com/projects/zig-a-zig-ah

https://electrolama.com/_assets/zzh-RevA-Draft1-010220.pdf

https://twitter.com/OmerK

sanyatuning, 2020-05-03:
working with zzh adapter

closed 2020-05-03 by sanyatuning