zigpy/zigpy-cc #9: Tag release versions and make PyPi release(s) of zigpy-cc?

Hedda, 2020-02-07:
A friendly reminder to tag an initial release version of zigpy-cc and make PyPi release of zigpy-cc

@sanyatuning I guess that you might not yet be ready to make your first release of this zigpy-cc library but I just wanted to add a friendly reminder here under issues for tracking purposes only, so there is no rush.

An initial PyPi release version tag (like for example v0.0.1) and then continues versions/releases of newer version if and when code is added or changed will make it much easier for projects like Home Assistant and Linux OS distributions like Hass.io and Raspbian or others to update and make use of each new versions of this library.

Version tagging and PyPi release(s) is something that is already available for the zigpy library as well as similar radio libraries that it depends on such as; bellows, zigpy-deconz, zigpy-xbee, and zigpy-zigate. My thinking is that the more people are testing this early on then more developers will eventually also get interested and will be able to help you out with the development of zigpy-cc

https://github.com/zigpy/zigpy/releases
https://github.com/zigpy/bellows/releases
https://github.com/zigpy/zigpy-xbee/releases
https://github.com/zigpy/zigpy-deconz/releases
https://github.com/doudz/zigpy-zigate/releases

https://pypi.org/project/zigpy
https://pypi.org/project/bellows
https://pypi.org/project/zigpy-xbee
https://pypi.org/project/zigpy-deconz
https://pypi.org/project/zigpy-zigate

Off-topic but maybe could talk @Adminiuga about the possibility of maybe having your zigpy-cc repository move the GitHub repo to the zigpy organization @ https://github.com/zigpy/ so that more than one person has access to all your zigpy related projects for collaboration? Hope that you could all come to an agreement that will make everyone happy with continuing with zigpy-cc

sanyatuning, 2020-02-08:
I've created a pypi package, you can find it in the readme.
@Adminiuga what do you think about the move?

Hedda, 2020-02-16:
I can see that you now also tagged v0.1.1 but looks like not pushed that to release to pypi?

Not sure if it possible to configure GitHub or PyPi to automatically push or pull releases?

https://github.com/sanyatuning/zigpy-cc/releases

https://pypi.org/project/zigpy-cc/

sanyatuning, 2020-02-17:
I think it is posible, just I have to write a little script which will update the version in [init.py](https://github.com/sanyatuning/zigpy-cc/blob/master/zigpy_cc/__init__.py)

sanyatuning, 2020-03-06:
done

closed 2020-03-06 by sanyatuning