zigpy/zigpy-cc #12: "No route found" with C26X2R1 on Z-Stack 3.x.0

puddly, 2020-02-11:
All `dataRequest` commands are immediately rejected with a `205` status code, a "no route found" error. The only solution I can find is to send out a route discovery broadcast message:

```Python
if obj.command == "dataRequest":
    # XXX: dataConfirm can arrive before dataRequest response

    payload = {
        "endpoint": obj.payload["destendpoint"],
        "transid": obj.payload['transid'],
    }
    waiter = self.wait_for(CommandType.AREQ, Subsystem.AF, "dataConfirm", payload)
    result = await waiter.wait()
    LOGGER.debug('Got a dataRequest response: %s', result)

    status = result.payload['status']

    if status != 0:
        # No route found
        if status == 205:
            BEACON_MAX_DEPTH = 0x0F
            DEF_NWK_RADIUS = (2 * BEACON_MAX_DEPTH)

            LOGGER.warning('Received a routing failure! Attempting to discover a new route before failing...')

            await self.request(Subsystem.ZDO, 'extRouteDisc', {
                'dstAddr': obj.payload['dstaddr'],
                'options': 0x01,
                'radius': DEF_NWK_RADIUS
            })

            await asyncio.sleep(5)

        raise CommandError(result.payload['status'], f'Received a bad dataRequest status: {result.payload["status"]}')
```

Only one request like this is necessary, though I'm not sure what the consequences are of enqueuing a bunch of these messages are (especially since all of the useful parts of the Z-Stack firmware are not open source).

sanyatuning, 2020-05-03:
Can't reproduce.
@puddly do you still have this problem?

puddly, 2020-05-03:
Occasionally. You have to enable route discovery or manually send a route discovery request.

puddly, 2020-05-03:
How many devices do you have in your network?