zigpy/zigpy-cc #16: Got unk_manufacturer unk_model while pairing Aqara Device

RockBomber, 2020-02-27:
Hello!
Thank You for zigpy-cc module! I glad to see it in new Home Assistant release.
But I have issue with pairing devices.
Maybe it is not related with You project, then say me please right project for report about the issue)

I have two zigbee devices: Aqara Water Leak sensor (https://www.zigbee2mqtt.io/devices/SJCGQ11LM.html) and Aqara Wireless Switch (https://www.zigbee2mqtt.io/devices/WXKG11LM.html)
And I have Home Assistant 0.106 with USB module **CC2351** runned under Ubuntu 18.04.4
I manually configured zha integration in configuration.yaml:
```
zha:
  radio_type: ti_cc
  usb_path: /dev/ttyACM0
  database_path: /config/zigbee.db
```
Firstly, each time after start Home Assistant I've got an error in homeassitant.log:
```
2020-02-27 23:19:55 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry /dev/ttyACM0 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 215, in async_setup
    hass, self
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 103, in async_setup_entry
    await zha_gateway.async_initialize()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 137, in async_initialize
    await self.application_controller.startup(auto_form=True)
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/zigbee/application.py", line 64, in startup
    self.version = await self._api.version()
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/api.py", line 206, in version
    version = await self._command(Subsystem.SYS, "version", {})
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/api.py", line 84, in _command
    return await self.request(subsystem, command, payload)
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/api.py", line 88, in request
    return await self.request_raw(obj, expectedStatus)
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/api.py", line 105, in request_raw
    result = await waiter.wait()
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/api.py", line 35, in wait
    return await asyncio.wait_for(self.future, self.timeout / 1000)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 449, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```
But I see the integration on integrations page.
Secondly, when I trying pairing any zigbee device, I see in homeassistant.log warnings like
```
2020-02-27 23:22:32 WARNING (MainThread) [homeassistant.components.zha.core.channels] [0x3c28:1:0x0000]: channel: 'async_configure' stage failed ex: 'NoneType' object has no attribute 'serialize'
2020-02-27 23:22:32 WARNING (MainThread) [homeassistant.components.zha.core.channels] [0x3c28:1:0x0006]: channel: 'async_configure' stage failed ex: 'NoneType' object has no attribute 'serialize'
```
And device added as unk_manufacturer unk_model
Creation of automations or scenarios not available for that device.

Can You help with this issue?
Thank You!)

puddly, 2020-02-27:
`await self._command(Subsystem.SYS, "version", {})` is the very first command that is sent. If it times out (which it shouldn't), then either you have the wrong serial port in your configuration, or your hardware has softlocked (also unlikely but possible).

@sanyatuning I would really hold off letting people use this module with a beginner-friendly system like Home Assistant. Currently there are a few unhandled network states (#12) and features (route re-discovery, NVRAM backups, concurrency limits, retrying, etc.) that can cause your entire Zigbee network to just completely "break" after a while. I'm slowly implementing those features in [my fork](https://github.com/puddly/zigpy-cc/tree/command-queue), but they're not quite ready yet.

RockBomber, 2020-02-28:
Serial port /dev/ttyACM0 is correct, because pairing of devices got a warning in logs, and new devices (unk_manufacturer unk_model) adds on integrations page.

How can I check for softlock?

puddly, 2020-02-28:
Can you enable [debug logging](https://www.home-assistant.io/integrations/logger/) for this package to see what's being sent and post the log? Something like:

```YAML
logger:
  logs:
    zigpy_cc: debug
```

RockBomber, 2020-02-28:
I'm sorry, I think I solved my problem) My mistake, I read inattentively zha doc page https://www.home-assistant.io/integrations/zha/
There is instruction to purge modemmanager. After that error on startup is gone, and I can pairing devices again)

closed 2020-02-28 by RockBomber