zigpy/zigpy-cc #18: devices disappears when restart HA

vMeph, 2020-03-02:
was wondering i have tried the new integration in home assistant, ZHA using the CC2531 radio_type: ti_cc, all works great, it discoveries de devices but then for some reason when you restart homeassistant all devices are gone from ZHA, any toughts?

Hedda, 2020-03-03:
Advice that you start troubleshooting/debugging from the top down instead of from the bottom up. Meaning, begin with checking logs in Home Assistant first, before moving on to zigpy and lastly zigpy-cc.

Suggest that post your logs in a new thread in Home Assistant community forum if you can't find an existing thread with the same issue.

https://community.home-assistant.io/c/configuration/zigbee/39

That is, make sure that the Home Assistant community exhausted their troubleshooting knowledge of ZHA before they point you here to posting issue with ZHA directly to a radio library like zigpy-cc.

Recommend that you enable DEBUG logging (first maybe only for homeassistant.core but later also for homeassistant.components.zha and zigpy_cc) and then in the home-assistant.log file look for zha. 

To enable debug logging either update logger configuration section in configuration.yaml or call logger.set_default_level service with {"level": "debug"} data

Check logger component configuration where you want something this in your configuration.yaml

 logger:
   default: info
   logs:
     asyncio: debug
     homeassistant.core: debug
     homeassistant.components.zha: debug
     zigpy: debug
     bellows: debug
     zigpy_xbee: debug
     zigpy_deconz: debug
     zigpy_zigate: debug
     zigpy_cc: debug

vMeph, 2020-03-05:
thanks for the guidance will look into it

closed 2020-03-05 by vMeph