zigpy/zigpy-cc #19: Logs full of errors

tehspede, 2020-03-03:
Hi,

My home assistant logs are full of the following
`2020-03-03 09:28:22 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback SerialTransport._read_ready()
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.7/site-packages/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/uart.py", line 140, in data_received
    self._api.data_received(frame)
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/api.py", line 187, in data_received
    waiter.set_result(obj)
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/api.py", line 38, in set_result
    self.future.set_result(result)
asyncio.base_futures.InvalidStateError: invalid state
2020-03-03 09:28:32 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 172, in request
    use_ieee=use_ieee,
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/zigbee/application.py", line 184, in request
    await self._api.request_raw(obj)
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/api.py", line 126, in request_raw
    result = await waiter.wait()
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/api.py", line 35, in wait
    return await asyncio.wait_for(self.future, self.timeout / 1000)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 449, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError`

Hedda, 2020-03-03:
Sorry, don't think anyone here is stilled at crystal ball reading. Please try to understand that you must always provide much more information about your setup and the steps you have taken before someone can even begin to help you. 

All I can say it that log to me looks very similar to https://github.com/sanyatuning/zigpy-cc/issues/16 in which there are some suggestions posted. Among them to go through the troubleshooting steps posted in https://www.home-assistant.io/integrations/zha/

tehspede, 2020-03-03:
Hi,

I am running home assistant in docker with the CC2531 being passed in to the container and have made sure that modemmanager is not installed on the host. I have the following devices set up in my network and for most of the time everything works fine and dandy.
[image]

However, I have noticed that after a while (I have not measured this so can't really say after how long, but I'm talking about hours) I start getting the errors mentions previously in my logs. I wen't ahead and turned on debugging and captured the logs before the errors occur: https://paste.ubuntu.com/p/XJrHQN6QHF/

Restarting home assistant gets rid of the errors. I just restarted home assistant so I'll try to keep an eye on the logs today

Some symptoms I'm having that might be related to this are:
- When the Trådfri lights are on without anybody touching anything, they might suddenly appear as off in home assistant.
- The Aqara wireless switches do not always toggle the lights, even though the automation is triggered in home assistant.
- When the above occurs, and the switch has been pressed multiple times to get the lights to toggle, they might randomly turn on or off a couple of minutes later.
- When turning lights on or off from home assistant, the state does not always reflect this. (e.g. turning a light on might still show them off in home assistant)

Hopefully this information helps and if theres anything more I can do to help I am more than glad to do so!

Hedda, 2020-03-03:
Just a feeling but sounds to me like you might have issues with signal strength (power) in your Zigbee mesh network. Suggest that you try to map your ZHA network topology to check LQI for all devices and specifically the route and routers for the devices that you have issues with. There are some addon tools for ZHA that can help with this so that you do not have to use a sniffer.

ZHA Map - Home Assistant can build ZHA network topology map using the zha-map project.
- https://github.com/zha-ng/zha-map

zha-network-visualization-card - a custom Lovelace element for visualizing the ZHA Zigbee network in Home Assistant.
- https://github.com/dmulcahey/zha-network-visualization-card

ZHA Network Card - a custom Lovelace card that displays ZHA network and device information in Home Assistant
- https://github.com/dmulcahey/zha-network-card

# **General tips for improving the range and communication within a Zigbee mesh network**

Add more routers between the problem devices and the next nearest router. In a Zigbee mesh network, each router will extend the range of the mesh network. As a rule of thumb, almost all permanently powered devices will serve as a router (however many Zigbee light bulbs are commonly excepted from that rule).

Adding a USB extension cable will move your USB stick away from the computer you've connected it to. . A USB extension cable of 1-meter and placing it away from any other radios/antennas can often reduce interference, and thus improve your range.

Add an external antenna to your coordinator, and position the antenna away from the computer. Next level of this is to add an antenna extension cable but be warned here that you should not use too long antenna extension cables as each meter added to decrease the signal strength.

Try different orientation of your Zigbee antenna (or your whole Zigbee adapter if it has an internal antenna). This is because the connection between the Zigbee radio of your "coordinator" and your other devices depends on the way the antenna oriented in space.

Ensure that any WiFi router/access point, WiFi USB stick, Z-Wave stick, or RF433 stick are as far away as you can get them (preferably at least one meter if you can). This will also reduce interference.

Check the WiFi channels in use, to see if your router/access point is now talking over your Zigbee mesh. If so change the channel of the WiFi access point and lock it. If you have to change the channel of your Zigbee mesh then you'll have to re-pair your devices.

Buy more powerful Zigbee radio hardware with better radio range, preferably with an external antenna.

If you're still stuck at that point, I advise that you head over the #zigbee channel on the [Home Assistant Discord server](https://discord.gg/c5DvZ4e), or the [Zigbee section of the community forum](https://community.home-assistant.io/c/configuration/zigbee).

Suggest that you then post your logs in a new thread in Home Assistant community forum if you can't find an existing thread with the same issue. Regardless of my advice that you start troubleshooting/debugging from the top down instead of from the bottom up. Meaning, begin with checking logs in Home Assistant ZHA first, before moving on to zigpy and lastly zigpy-cc. 

https://community.home-assistant.io/c/configuration/zigbee/39

That is, make sure that the Home Assistant community exhausted their troubleshooting knowledge of ZHA before they point you here to posting issue with ZHA directly to a radio library like zigpy-cc.

Recommend that you enable DEBUG logging (first maybe only for homeassistant.core but later also for homeassistant.components.zha and zigpy_cc) and then in the home-assistant.log file look for zha.

To enable debug logging either update logger configuration section in configuration.yaml or call logger.set_default_level service with {"level": "debug"} data

Check logger component configuration where you want something this in your configuration.yaml

logger:
default: info
logs:
asyncio: debug
homeassistant.core: debug
homeassistant.components.zha: debug
zigpy: debug
bellows: debug
zigpy_xbee: debug
zigpy_deconz: debug
zigpy_zigate: debug
zigpy_cc: debug

tehspede, 2020-03-03:
Hey,

I can try investigating with zha-map, but this setup worked perfectly with zigbee2mqtt before and to me it sounds weird that a home assistant restart would resolve signal strength issues.

Hedda, 2020-03-03:
The router routes do not necessarily need to be the same for your current ZHA network topology as they were in your previous Zigbee2mqtt network topology, meaning the network topology might be different even if all the devices was migrated over. 

ZHA network topology will depend on which order new devices were added as well as if ZHA actually see and use all router capable devices like routers or not. Yes normally the first mains-powered devices will be used as routers but it is no guarantee.

Optimally you should today actively build your network topology by first adding all your routers one-by-one to make sure that the ones physically closest to the coordinator will act as direct children routers before adding additional devices.

These are just best-practices and might in fact not reflect your issues, but I still suggest to follow those general recommendations. Again, your issue might not be with this zigpy-cc library so you should aim to first get support from the Home Assistant community.

tehspede, 2020-03-03:
I actually did build the network by adding the routers starting from the closest and adding each device through the "Add Devices" from the closest router.

LEJOUI, 2020-03-08:
I’m also having this same errors in my logs. Also ran fine fine on Z2M. The errors seem, not entirely sure, to keep coming from the tradfri bulbs on my end.

sanyatuning, 2020-03-09:
I created a new release (v0.2.2) with potential fixes.
Please test it if you can.
If you use "hass.io" the update is not easy.

LEJOUI, 2020-03-09:
I can ssh into the docker container and replace the zigby_cc with the new release. Do i have to change something else if I want to test it?

sanyatuning, 2020-03-09:
It should work, I think
I am not a python guru :)

LEJOUI, 2020-03-09:
Thanks. I did replace the folder, but now I'm getting new errors:

[image]

Any ideas?

sanyatuning, 2020-03-26:
these are debug messages, log levels fixed

closed 2020-03-26 by sanyatuning