zigpy/zigpy-cc #21: zigpy_cc is throwing an exception when decoding a received frame from the serial port

macros0, 2020-03-05:
https://github.com/zha-ng/zha-map/issues/15

aneisch, 2020-03-06:
I am having the same:

```File "/usr/local/lib/python3.7/site-packages/zigpy_cc/zpi_object.py", line 76, in from_unpi_frame
payload = cls.read_parameters(frame.data, parameters)
File "/usr/local/lib/python3.7/site-packages/zigpy_cc/zpi_object.py", line 149, in read_parameters
res[name] = buffalo.read_parameter(type, options)
File "/usr/local/lib/python3.7/site-packages/zigpy_cc/buffalo.py", line 69, in read_parameter
elif type == ParameterType.LIST_UINT32:
File "/usr/local/lib/python3.7/enum.py", line 349, in getattr
raise AttributeError(name) from None
AttributeError: LIST_UINT32```

sanyatuning, 2020-03-06:
Thanks for the report. I will fix it this week.

sanyatuning, 2020-03-06:
self note:
there are no LIST_UINT32 in ParameterType 
and looks like buffaloznp class is not ported

sanyatuning, 2020-03-20:
@macros0 I fixed this
Can you upgrade zigpy-cc in your installation to test it?

ps: It will be available in HA 0.108.

sanyatuning, 2020-03-20:
I added some info to the readme about how you can test the new release

macros0, 2020-03-23:
Hassio 107.5, no start hassio-addons-development

[Info] Remove old deps
[Info] Install pypi modules into deps
You are using pip version 19.0.2, however version 20.0.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
[Info] done

LEJOUI, 2020-03-23:
I did the upgrade to 0.3.0 and it throws this error when running a network scan:

`Traceback (most recent call last):
  File "/usr/local/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.7/site-packages/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/config/deps/lib/python3.7/site-packages/zigpy_cc/uart.py", line 144, in data_received
    self._api.data_received(frame)
  File "/config/deps/lib/python3.7/site-packages/zigpy_cc/api.py", line 201, in data_received
    raise e
  File "/config/deps/lib/python3.7/site-packages/zigpy_cc/api.py", line 198, in data_received
    obj = ZpiObject.from_unpi_frame(frame)
  File "/config/deps/lib/python3.7/site-packages/zigpy_cc/zpi_object.py", line 76, in from_unpi_frame
    payload = cls.read_parameters(frame.data, parameters)
  File "/config/deps/lib/python3.7/site-packages/zigpy_cc/zpi_object.py", line 149, in read_parameters
    res[name] = buffalo.read_parameter(type, options)
  File "/config/deps/lib/python3.7/site-packages/zigpy_cc/buffalo.py", line 73, in read_parameter
    raise TODO("read type %d", type)
zigpy_cc.exception.TODO: ('read type %d', <ParameterType.LIST_NEIGHBOR_LQI: 15>)`

sanyatuning, 2020-03-23:
i will look into this

Sebclem, 2020-04-17:
Hi !
I still have the TODO exception with 0.3.1. (HA 0.108.5) but with a different line (41) : 
```2020-04-17 14:59:51 DEBUG (MainThread) [zigpy_cc.zigbee.application] request (0x0000, 0, <ZDOCmd.Mgmt_Lqi_req: 49>, 0, 0, 13, b'\r\x00', True, False)
2020-04-17 14:59:51 DEBUG (MainThread) [zigpy_cc.api] waiting for 13 mgmtLqiReq
2020-04-17 14:59:51 DEBUG (MainThread) [zigpy_cc.api] --> SREQ ZDO mgmtLqiReq tsn: 13 {'dstaddr': 0x0000, 'startindex': 0}
2020-04-17 14:59:51 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x03%1\x00\x00\x00\x17'
2020-04-17 14:59:51 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.ZDO command_id=49 data=b'\x00' length=1 fcs=85>
2020-04-17 14:59:51 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP ZDO mgmtLqiReq tsn: None {'status': 0}
2020-04-17 14:59:51 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=177 data=b'\x00\x00\x00\x01\x00\x01\xdd\xdd\xdd\xdd\xdd\xdd\xdd\xdd\x0f~O\x04\x00\x8d\x15\x00|\x1b\x12\x02\x01j' length=28 fcs=86>
2020-04-17 14:59:51 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO mgmtLqiRsp tsn: 13 {'srcaddr': 0x0000, 'status': 0, 'neighbortableentries': 1, 'startindex': 0, 'neighborlqilistcount': 1, 'neighborlqilist': [{'extPanId': dd:dd:dd:dd:dd:dd:dd:dd, 'extAddr': 00:15:8d:00:04:4f:7e:0f, 'nwkAddr': 0x1b7c, 'deviceType': 2, 'rxOnWhenIdle': 0, 'relationship': 1, 'permitJoin': 2, 'depth': 1, 'lqi': 106}]}
2020-04-17 14:59:51 WARNING (MainThread) [zigpy_cc.exception] Not implemented: write ParameterType.LIST_NEIGHBOR_LQI
2020-04-17 14:59:51 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback SerialTransport._read_ready()
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.7/site-packages/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/uart.py", line 144, in data_received
    self._api.data_received(frame)
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/api.py", line 219, in data_received
    self._app.handle_znp(obj)
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/zigbee/application.py", line 278, in handle_znp
    frame = obj.to_unpi_frame()
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/zpi_object.py", line 55, in to_unpi_frame
    data.write_parameter(p["parameterType"], value, {})
  File "/usr/local/lib/python3.7/site-packages/zigpy_cc/buffalo.py", line 41, in write_parameter
    raise TODO("write %s", ParameterType(type))
zigpy_cc.exception.TODO: ('write %s', <ParameterType.LIST_NEIGHBOR_LQI: 15>) 
```
It's just after a zha_map.scan_now call.

sanyatuning, 2020-04-18:
Thanks for the report.
I will try zha_map to fix all errors at once.

sanyatuning, 2020-05-03:
fixed, release soon

sanyatuning, 2020-05-12:
released in 0.3.2 and 0.4.2

commit bb5e69ba (2020-03-25): feature: handle mgmtLqiRsp, fixes #21
closed 2020-05-12 by sanyatuning