zigpy/zigpy-cc #22: [REQUEST] Capability to handle serial-port disconnects and reconnects

Hedda, 2020-03-06:
It is possible to implement some type of handling of serial port disconnects and reconnects?

The main reason for that is when using ser2net and socat to use serial device over the network.

Problem is if the network temporarily goes down then it can not reconnect to recover the connection. 

This is a relatively common networked setup if don't want the radio close to Home Assistant, see: 

https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

Another problem scenario is if the user accidentally unplugs the Zigbee USB adapter when running, as seen in related issue: 

https://github.com/sanyatuning/zigpy-cc/issues/15

References:

https://community.home-assistant.io/t/add-network-protocol-support-to-zigbee-and-zwave-integrations/136056

https://github.com/Danielhiversen/pyRFXtrx/pull/85

https://github.com/zigpy/bellows/issues/183

https://github.com/zigpy/zigpy/issues/297

Hedda, 2020-03-06:
Please also see the somewhat related request for auto USB/serial-port & radio-type discovery in https://github.com/sanyatuning/zigpy-cc/issues/20

sanyatuning, 2020-03-07:
@Hedda can you test this function with the new release?

closed 2020-04-28 by sanyatuning