zigpy/zigpy-cc #23: [SUGGESTION] Team up with Electrolama to beta-test the "Zig-a-zig-ah!" CC2652 adapter

Hedda, 2020-03-09:
@sanyatuning Can I suggest that you try to reach out to Omer Kilic a.k.a. @omerk at Electrolama about teaming up to beta-test his "Zig-a-zig-ah!" CC2652 adapter with zigpy-cc for for ZHA in Home Assistant?  

https://electrolama.com/projects/zig-a-zig-ah/

https://github.com/electrolama/zig-a-zig-ah

As I understand both zig-a-zig-ah and zigpy-cc are only spare-time hobby projects but still I think that teaming up might give both projects some positive recognition in the Home Assistant community which might attract more developers and willing alpha/beta-testers from the community to come help out with both these projects which in the long run could benefit all users wanting ZHA to become more mature and working with more hardware than today.

https://www.home-assistant.io/components/zha/

Would also be nice to see collaboration with the guys working on zigpy-znp as an alternative

https://github.com/zha-ng/zigpy-znp

Hedda, 2020-03-11:
@sanyatuning Checkout related https://github.com/electrolama/zig-a-zig-ah/issues/5 where @omerk of electrolama.com express a willingness to collaborate and could provide some zig-a-zig-ah adapter prototypes to a few zigpy and ZHA developers to help the development of TI CC2652 support. You can send him an e-mail if you are interested.

Hedda, 2020-03-23:
@sanyatuning not sure if you already sent Omer a mail regarding getting a zig-a-zig-ah (TI CC2652 USB-stick) prototype, but FYI, Koenkk from developed zigbee-shepherd just wrote that he has received a zig-a-zig-ah prototype from Omer and it seems to work great so far with zigbee2mqtt https://github.com/Koenkk/zigbee2mqtt/issues/1429#issuecomment-602485214

sanyatuning, 2020-03-23:
Nice news!
I mailed him now.

sanyatuning, 2020-04-28:
I am waiting for the prototype! 🎉

sanyatuning, 2020-05-03:
I received the prototype, flashed it with zStack3x0.
It is working nicely.

closed 2020-05-03 by sanyatuning