zigpy/zigpy-cc #26: Can anyone confirm out-of-the-box compatibility with ITEAD CC2531 USB-dongle?

Hedda, 2020-03-25:
ITEAD has started selling this $4 US-dollar CC2531 USB-dongle pre-flashed with a coordinator firmware which they say is plug-and-play compatible with Zigbee2mqtt out-of-the-box. So wondering if anyone has got one and confirm the preflashed firmware is also compatible with zigpy-cc too? 

- https://www.itead.cc/cc2531-usb-dongle.html

@sanyatuning  I'm sure that ITEAD if contact would be willing to donate a bunch of them to developers of well-known open-source projects like Home Assistant and its ZHA integration with Zigpy by affiliation. 

Regardless, might it good enough for most devs even if it is a model with an integrated antenna? Only me thinking out loud that it would be good if more Home Assistant developers had this hardware, might make them more willing to help with zigpy support for these cheap TI CC dongles?

sanyatuning, 2020-03-25:
I think it is compatible.

Hedda, 2020-06-04:
FYI, more than one person now confirmed that it works but ships with an old firmware from "20180507"

- https://github.com/zha-ng/zigpy-znp/issues/14

- https://community.home-assistant.io/t/itead-of-sonoff-fame-now-sells-a-very-inexpensive-zigbee-usb-dongle-that-should-work-out-of-the-box-with-both-zigbee2mqtt-and-the-zha-integration/181085/5

closed 2020-06-04 by sanyatuning