zigpy/zigpy-cc #27: cp2102 + cc2530

levran, 2020-03-26:
Hi!
Please add to HA configiration for ZHA rtscts option.
I tried to add cc2530 to HA/ZHA with no success.
My config:
zha:
  radio_type: ti_cc
  usb_path: /dev/ttyUSB0
  baudrate: 115200

At the same time, on the same host in the zigbee2mqtt (container), the cc2530+cp2102 stick works great if added the parameter in config rtscts: false.

Thank you!!!
BR, Vlad.

sanyatuning, 2020-03-27:
Can you send your logs?

Maybe I can auto detect it.

levran, 2020-03-27:
Hi!
Thank you for question!

There is logs from HA container.

Best regards, Vladislav.

2020-03-27 07:51:05 INFO (MainThread) [homeassistant.setup] Setup of domain
map took 0.0 seconds.,
2020-03-27 07:51:05 INFO (MainThread) [homeassistant.setup] Setting up
logbook,
2020-03-27 07:51:05 INFO (MainThread) [homeassistant.setup] Setup of domain
logbook took 0.0 seconds.,
2020-03-27 07:51:05 INFO (SyncWorker_5) [homeassistant.loader] Loaded
notify from homeassistant.components.notify,
2020-03-27 07:51:07 INFO (MainThread) [homeassistant.setup] Setting up
default_config,
2020-03-27 07:51:07 INFO (MainThread) [homeassistant.setup] Setting up
notify,
2020-03-27 07:51:07 INFO (MainThread) [homeassistant.setup] Setup of domain
notify took 0.0 seconds.,
2020-03-27 07:51:07 INFO (MainThread) [homeassistant.components.notify]
Setting up notify.mobile_app,
2020-03-27 07:51:07 INFO (SyncWorker_17) [homeassistant.loader] Loaded
stream from homeassistant.components.stream,
2020-03-27 07:51:07 INFO (SyncWorker_19) [homeassistant.loader] Loaded hue
from homeassistant.components.hue,
2020-03-27 07:51:07 INFO (MainThread) [homeassistant.setup] Setting up
stream,
2020-03-27 07:51:07 INFO (MainThread) [homeassistant.setup] Setup of domain
stream took 0.0 seconds.,
2020-03-27 07:51:07 INFO (MainThread) [homeassistant.setup] Setup of domain
default_config took 0.4 seconds.,
2020-03-27 07:51:09 ERROR (MainThread)
[homeassistant.components.zha.core.gateway] Couldn't start EZSP coordinator,
Traceback (most recent call last):,
  File
"/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line
142, in async_initialize,
    res = await self.application_controller.startup(auto_form=True),
  File
"/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py",
line 135, in startup,
    await self.initialize(),
  File
"/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py",
line 72, in initialize,
    await e.reset(),
  File "/usr/local/lib/python3.7/site-packages/bellows/ezsp.py", line 57,
in reset,
    await self._gw.reset(),
  File "/usr/local/lib/python3.7/site-packages/bellows/uart.py", line 222,
in reset,
    return await asyncio.wait_for(self._reset_future,
timeout=RESET_TIMEOUT),
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 449, in wait_for,
    raise futures.TimeoutError(),
concurrent.futures._base.TimeoutError,
2020-03-27 07:51:09 WARNING (MainThread) [homeassistant.config_entries]
Config entry for zha not ready yet. Retrying in 5 seconds.,
2020-03-27 07:51:09 INFO (SyncWorker_13) [homeassistant.loader] Loaded
google_translate from homeassistant.components.google_translate,
2020-03-27 07:51:09 INFO (MainThread) [homeassistant.setup] Setting up tts,
2020-03-27 07:51:09 INFO (MainThread) [homeassistant.setup] Setup of domain
tts took 0.0 seconds.,
2020-03-27 07:51:09 INFO (MainThread) [homeassistant.bootstrap] Home
Assistant initialized in 18.15s,
2020-03-27 07:51:09 INFO (MainThread) [homeassistant.core] Starting Home
Assistant,
2020-03-27 07:51:09 INFO (MainThread) [homeassistant.core] Timer:starting,
[... 85 lines cut ...]
Traceback (most recent call last):,
  File
"/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line
142, in async_initialize,
    res = await self.application_controller.startup(auto_form=True),
  File
"/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py",
line 135, in startup,
    await self.initialize(),
  File
"/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py",
line 72, in initialize,
    await e.reset(),
  File "/usr/local/lib/python3.7/site-packages/bellows/ezsp.py", line 57,
in reset,
    await self._gw.reset(),
  File "/usr/local/lib/python3.7/site-packages/bellows/uart.py", line 222,
in reset,
    return await asyncio.wait_for(self._reset_future,
timeout=RESET_TIMEOUT),
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 449, in wait_for,
    raise futures.TimeoutError(),
concurrent.futures._base.TimeoutError,
2020-03-27 07:52:48 WARNING (MainThread) [homeassistant.config_entries]
Config entry for zha not ready yet. Retrying in 80 seconds.,

sanyatuning, 2020-03-27:
Sorry but this log sad you set radio type to ezsp instead of ti_cc.
Can you test it again?

levran, 2020-03-27:
**on host** machine Ubuntu:
ls -l /dev/serial/by-id/
usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -> ../../ttyUSB0

there my **configiration.yaml**
zha:
  radio_type: ti_cc
  usb_path: /dev/ttyUSB0
  baudrate: 115200
  enable_quirks: false

if i use HA interface to add ZHA, see pictures
[image]
[image]

levran, 2020-03-27:
I understand that this is completely off topic of this problem. But I also tried ConbeeII connect to the HA through the ZHA and it also worked. And even all paired devices became visible.

Nickolay40, 2020-03-28:
Hello, I have same issue, but use just coordinator F/W(without mqtt). For example, in iobroker driver this configuration working fine.

sanyatuning, 2020-03-30:
maybe fixed in commit 58c445e5ccfe00c27d1b2bcca0a923cfe2055840

sanyatuning, 2020-03-30:
pls test with release v0.3.1

levran, 2020-03-30:
I do not know how otherwise. I edited the uart.py in the docker container (HA 0.107.7). 
The result is in the log.
I tried to attach a xiaomi door sensor, it is clear that there is a reaction. 
But as a separate device, the sensor is not visible.
[home-assistant.log](https://github.com/zigpy/zigpy-cc/files/4404193/home-assistant.log)

sanyatuning, 2020-03-30:
[which](https://www.zigbee2mqtt.io/information/supported_devices.html#xiaomi) door sensor you are using?
Can you pair it?
I only see timeouts in your log.

levran, 2020-03-30:
Xiaomi MCCGQ01LM
When i search new device by "add device" i see in log window below new lines, when i open and close window sensor.
Tomorrow i'll write exact lines from log.

levran, 2020-03-31:
Ignoring message (b'18320a00001001') on cluster 6: unknown endpoint or cluster id: 1
[0xfda0] Requesting 'Node Descriptor'
[0xfda0] Extending timeout for 0x41 request
Ignoring message (b'18330a00001000') on cluster 6: unknown endpoint or cluster id: 1 <---- *** this is sensor close action ***
Ignoring message (b'18340a00001001') on cluster 6: unknown endpoint or cluster id: 1 <---- *** this is sensor open action ***
[0xfda0:zdo] ZDO request 0x8002: [<Status.SUCCESS: 0>, 0xfda0, <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4151 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=0 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>]
[0xfda0:zdo] Unsupported ZDO request:0x8002
Ignoring message (b'1c3412350a02ff4c0600100121d10b21a813240400000000214000205e') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18360a00001000') on cluster 6: unknown endpoint or cluster id: 1  <---- *** this is sensor close action ***
Ignoring message (b'18370a00001001') on cluster 6: unknown endpoint or cluster id: 1  <---- *** this is sensor open action ***
Ignoring message (b'18380a00001000') on cluster 6: unknown endpoint or cluster id: 1  <---- *** this is sensor close action ***
Ignoring message (b'18390a00001001') on cluster 6: unknown endpoint or cluster id: 1  <---- *** this is sensor open action ***
Ignoring message (b'183a0a00001000') on cluster 6: unknown endpoint or cluster id: 1  <---- *** this is sensor close action ***
Ignoring message (b'183b0a00001001') on cluster 6: unknown endpoint or cluster id: 1  <---- *** this is sensor open action ***
[0xfda0] Extending timeout for 0x43 request

levran, 2020-04-10:
I tried new version HA 0.108.2, cc2530 throw cp2102 works.

closed 2020-04-10 by levran