zigpy/zigpy-cc #31: cc2530 + QBKG03LM. only shows status, no control.

levran, 2020-04-10:
the switch QBKG03LM is successfully paired with cc2530 + cp2102, is shown in ZHA. When physically turning on / off, the status changes in the HA (0,108,2)  interface. But changing the status from the HA interface does not lead to a physical switch. In addition, after a few seconds in the interface HA, the switch returns to its previous state.

**pairing part:**
Device 0x32ab (00:15:8d:00:01:4a:95:d6) joined the network
Device 00:15:8d:00:01:4a:95:d6 changed id (0x930e => 0x32ab)
[0x32ab:zdo] ZDO request ZDOCmd.Device_annce: [0x32ab, 00:15:8d:00:01:4a:95:d6, 132]
Checking quirks for LUMI lumi.ctrl_neutral2 (00:15:8d:00:01:4a:95:d6)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 5, 6, 8}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 2, 3, 4, 5, 6, 8}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 5, 6, 8}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 5, 6, 8}
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1, 2, 3, 4, 5, 6, 8}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 5, 6, 8}
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1, 2, 3, 4, 5, 6, 8}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1, 2, 3, 4, 5, 6, 8}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 5, 6, 8}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 5, 6, 8}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2, 3, 4, 5, 6, 8}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2, 3, 4, 5, 6, 8}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 5, 6, 8}
device - 0x32ab:00:15:8d:00:01:4a:95:d6 entering async_device_initialized - is_new_join: False
device - 0x32ab:00:15:8d:00:01:4a:95:d6 has been reset and re-added or its nwk address changed
skipping discovery for previously discovered device - 0x32ab:00:15:8d:00:01:4a:95:d6
[0x32ab](lumi.ctrl_neutral2): started configuration
[0x32ab:ZDO](lumi.ctrl_neutral2): 'async_configure' stage succeeded
[0x32ab:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=0 command_id=Command.Report_Attributes>
[0x32ab:1:0x0000] ZCL request 0x000a: [[<Attribute attrid=5 value=<TypeValue type=CharacterString, value=lumi.ctrl_neutral2>>, <Attribute attrid=1 value=<TypeValue type=uint8_t, value=1>>]]
[0x32ab:1:0x0000] Attribute report received: model=lumi.ctrl_neutral2, app_version=1
[0x32ab:2:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=1 command_id=Command.Report_Attributes>
[0x32ab:2:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.false>>, <Attribute attrid=61440 value=<TypeValue type=uint32_t, value=50331392>>]]
[0x32ab:2:0x0006] Attribute report received: on_off=Bool.false, 61440=50331392
[0x32ab:3:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=2 command_id=Command.Report_Attributes>
[0x32ab:3:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.false>>, <Attribute attrid=61440 value=<TypeValue type=uint32_t, value=50331392>>]]
[0x32ab:3:0x0006] Attribute report received: on_off=Bool.false, 61440=50331392
[0x32ab:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x32ab:1:0x0000]: finished channel configuration
[0x32ab:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=71 command_id=Command.Read_Attributes_rsp>
[0x32ab:1:0x0000]: initializing channel: from_cache: False
[0x32ab:1:0x0019]: bound 'ota' cluster: Status.SUCCESS
[0x32ab:1:0x0019]: finished channel configuration
[0x32ab:1:0x0000]: 'async_configure' stage succeeded
[0x32ab:1:0x0019]: 'async_configure' stage succeeded
[0x32ab:2:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0x32ab:3:0x0006]: Failed to bind 'on_off' cluster: 
[0x32ab:4:0x0012]: Failed to bind 'multistate_input' cluster: 
[0x32ab:4:0x0012]: finished channel configuration
[0x32ab:zdo] ZDO request ZDOCmd.Bind_rsp: [<Status.SUCCESS: 0>]
[0x32ab:zdo] Unsupported ZDO request:ZDOCmd.Bind_rsp
[0x32ab:2:0x0006]: failed to set reporting for 'on_off' attr on 'on_off' cluster: 
[0x32ab:2:0x0006]: finished channel configuration
[0x32ab:2:0x0006]: 'async_configure' stage succeeded
[0x32ab:zdo] ZDO request ZDOCmd.Bind_rsp: [<Status.TABLE_FULL: 140>]
[0x32ab:zdo] Unsupported ZDO request:ZDOCmd.Bind_rsp
[0x32ab:4:0x0006]: Failed to bind 'on_off' cluster: 
[0x32ab:3:0x0006]: failed to set reporting for 'on_off' attr on 'on_off' cluster: 
[0x32ab:3:0x0006]: finished channel configuration
[0x32ab:3:0x0006]: 'async_configure' stage succeeded
[... 21 lines cut ...]
[0x32ab:8:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=99 command_id=Command.Configure_Reporting_rsp>
[0x32ab:8:0x000c]: reporting 'present_value' attr on 'analog_input' cluster: 30/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.UNREPORTABLE_ATTRIBUTE direction=ReportingDirection.SendReports attrid=85>, <ConfigureReportingResponseRecord status=Status.INVALID_DATA_TYPE direction=ReportingDirection.SendReports attrid=32768>]]'
[0x32ab:8:0x000c]: finished channel configuration
[0x32ab:8:0x000c]: 'async_configure' stage succeeded
[0x32ab:6:0x0006]: failed to set reporting for 'on_off' attr on 'on_off' cluster: 
[0x32ab:6:0x0006]: finished channel configuration
[0x32ab:6:0x0006]: 'async_configure' stage succeeded
[0x32ab:6:0x0012]: 'async_configure' stage succeeded
[0x32ab](lumi.ctrl_neutral2): completed configuration
[0x32ab](lumi.ctrl_neutral2): stored in registry: ZhaDeviceEntry(name='LUMI lumi.ctrl_neutral2', ieee='00:15:8d:00:01:4a:95:d6', last_seen=1586554352.0505073)
[0x32ab:1:0x0003] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=101 command_id=Command.Default_Response>
[0x32ab:1:0x0003]: executed 'trigger_effect' command with args: '(2, 0)' kwargs: '{}' result: [64, <Status.UNSUP_CLUSTER_COMMAND: 129>]
[0x32ab:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=3 command_id=Command.Read_Attributes>
[0x32ab:1:0x000a] ZCL request 0x0000: [[0]]

**Log then i try to turn on light:**
2020-04-10 21:36:14 INFO (MainThread) [zigpy_cc.uart] Bytes received: b'\x03D\x80\x00\x03'
2020-04-10 21:36:14 INFO (MainThread) [zigpy_cc.uart] Bytes received: b'\xfe'
2020-04-10 21:36:14 INFO (MainThread) [zigpy_cc.uart] Bytes received: b'\x01d'
2020-04-10 21:36:14 INFO (MainThread) [zigpy_cc.uart] Bytes received: b'\xfe'
2020-04-10 21:36:15 INFO (MainThread) [zigpy_cc.uart] Bytes received: b'\xfe'
2020-04-10 21:36:15 INFO (MainThread) [zigpy_cc.uart] Bytes received: b'\xfe'

BR,
Vladislav

levran, 2020-04-10:
it looks like the lack of Quirk for two-key switch.
[image]

sanyatuning, 2020-04-14:
I will try to create a quirk for that too.
Note: Zhaquirks has a separate repo https://github.com/zigpy/zha-device-handlers

levran, 2020-04-15:
moved to  https://github.com/zigpy/zha-device-handlers

sanyatuning, 2020-05-03:
https://github.com/zigpy/zha-device-handlers/pull/349

closed 2020-05-03 by sanyatuning