zigpy/zigpy-cc #32: [HW RECOMMENDATION] CC1352P-2 verses CC26X2R1 verses CC2652-based Zig-a-zig-ah?

Hedda, 2020-04-14:
What hardware would you recommend someone to buy for Z-Stack 3.x Zigbee coordinator firmware?

- https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

Personally I am low on funds right now so thinking of just waiting for Electrolama to start selling its CC2652-based [Zig-a-zig-ah USB-stick](https://electrolama.com/projects/zig-a-zig-ah/) but what hardware would you recommend today to someone who has funds and is not willing to wait?

It looks like most people in the Zigbee2mqtt community who has made the lead is currently recommending the slightly more expensive CC13x2 based LAUNCHXL-CC1352P-2 development kit over the CC26x2 based LAUNCHXL-CC26X2R1 development kit for a number of reasons, though mainly because CC1352P-2 has a built-in RF-amplifier (PA/LNA) as well as an SMA-connector for an external antenna and therefore has a much better link quality (people are even complaining that LAUNCHXL-CC26X2R1 has worse link quality than a CC2531 with an external antenna).

- https://github.com/Koenkk/zigbee2mqtt/issues/1429
  - https://zigbee2mqtt.discourse.group/t/cc1352p-ore-cc26x2r/488/2

[LAUNCHXL-CC1352P-2 ](http://www.ti.com/tool/LAUNCHXL-CC1352P)
  - +1: Has a built-in PA/LNA for potentially better range (assuming Z2M will support enablement - enhancement is raised). RF design includes a shield.
  - +1: Has a built-in SMA antenna connector (although it's still necessary to de/re-solder a tiny SMD capacitor switch from PCB antenna to SMA. Should provide an increase in range.
  - +1: Can support sub 1GHz - opening the possibility for 868MHz (EU) and 915MHz (NA) Zigbee for better range (technically possible but no idea how many end devices support it) and possibly other applications in the future… IMO this “benefit” should not be a consideration at the point.
  - -1: Bigger PCB (~2cm) than the CC26X2R1
  - -1: A little ($10) more expensive than the CC26X2R1
There are two versions with different RF suitability - The P-2 version favours 2.4GHz rather than sub 1GHz and this would be the recommended version to obtain for Z2M
  - +1: STL for 3D printed case here: https://www.thingiverse.com/thing:3928171

[LAUNCHXL-CC26X2R1](http://www.ti.com/tool/LAUNCHXL-CC26X2R1)
  - +1: Seems to be the preferred board on this thread (maybe we need a poll?)
  - +1: Slightly smaller PCB than the CC1352P
  - +1: A little cheaper than the CC1352P and more sellers than just TI.
  - +1: There is an STL for 3D printed case: https://www.thingiverse.com/thing:3305478
  - +1: Another STL for 3D printed case here: https://www.thingiverse.com/thing:3928171
  - -1: Theoretically less range than CC1352P due to lack of RF frontend
  - -1: Possibility to use external antenna (if de/re-solder the tiny SMD capacitor) but that requires a JSC-SMA antenna adaptor which is not cheap (~20 euros)
  - -1: No support for sub-1GHz but there is no clear advantage of this currently

Other notes:
  - There does not yet appear to be any more economical options from online China retailers like Aliexpress, etc. for either of these.
  - Both have built-in JTAG programmer so no need for an external programmer

sanyatuning, 2020-05-03:
I received a ZZH adapter for testing and it working nicely for me

Hedda, 2020-05-04:
@sanyatuning What are LQI and RSSI values reported by your different CCxxxx devices and sensors?

FYI, I read that OmerK has been or is looking after manufacturing partner(s) for the Electrolama Zig-a-zig-ah (zzh) adapter for a production version. Read that he was/is considering both crowdfunding sites, such as Crowd Supply and Tindie, as well as companies with already established supply-chains, like ITead (of Sonoff fame).

Anyway, current Zig-a-zig-ah prototype is also lacking PA (Power Amplifier) as it doesn't use CC2652P

https://electrolama.com/projects/zig-a-zig-ah/

  "Also thinking of a second variant that uses CC2652P, sadly not pin compatible with CC2652R (or CC2652RB) so this will have be a new board spin. "P" part has a built-in PA (but no LNA) and should provide better range."

https://www.ti.com/product/CC2652P

C2652R look to have practically the same spec as CC2652P but without integrated PA (Power Amplifier)

https://www.ti.com/product/CC2652R