zigpy/zigpy-cc #42: Option to disable LED on cc2531

Yevgenium, 2020-04-17:
Is it possible to add the option to disable LED on cc2531?

sanyatuning, 2020-04-18:
Possible in the near future.
There's an upcoming feature in zigy which will allow this.

Yevgenium, 2020-04-19:
For me, this is not the most critical. I am ready to wait for it to be implemented in zigpy.

Thanks!

Hedda, 2020-04-20:
Copied this request to upstream zigpy library in https://github.com/zigpy/zigpy/issues/384 as you are dependent on it being there first? 

I understand this is also related to https://github.com/zigpy/zigpy/issues/381 as another example of an optional extension?

Hedda, 2020-05-13:
Please also follow/see request discussion for zigpy-znp in https://github.com/zha-ng/zigpy-znp/issues/11

commit 38556957 (2020-05-13): Turn on led only while pairing, Resolve #42
closed 2020-05-13 by sanyatuning