zigpy/zigpy-cc #43: Cannot add new devices to ZHA in Home assistant

stevetz, 2020-05-01:
I posted this issue in zigpy/zigpy and home-assistant/core but have been asked to share it here too.
[https://github.com/home-assistant/core/issues/35004](https://github.com/home-assistant/core/issues/35004)
[https://github.com/zigpy/zigpy/issues/398](https://github.com/zigpy/zigpy/issues/398)

I'm trying to add them to my ZHA on Home Assistant - hassio. It has a bunch of Tradfri and Aqara devices already paired - via a cc2531 (passed through to the hassio vm on a Proxmox node). Running core 0.109.1 (upgraded from 107.7 which had the same issues, no change), supervisor fully updated etc.

Every time I try to pair a new device it fails to find anything. and this error is spat out:

`WARNING (MainThread) [zigpy_cc.zigbee.application] missing tsn from mgmtPermitJoinRsp, maybe not a reply`

sanyatuning, 2020-05-02:
This message is normal (I will remove it)
What device do you want to pair?
pls provide debug logs

HA config to get debug logs.
```
logs:
  homeassistant.components.zha: debug
  zigpy: debug
  zigpy_cc: debug
```

stevetz, 2020-05-02:
I am trying to pair Ikea Tradfri bulbs and control outlets, and aqara temp sensors. Since my last message I removed ZHA integration, deleted the zigbee.db, rebooted the vm and tried to start from scratch... Now it wont add the same devices which were initially paired - 3 days ago (HA 0.107.5, since upgraded to 0.109.2).

Here is the log from just now, trying to pair new devices:
````
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.zigbee.application] broadcast (0, 54, 0, 0, 0, 0, 1, b'\x01<\x00', <BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>)
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.api] --> SREQ ZDO mgmtPermitJoinReq tsn: 1 {'addrmode': 15, 'dstaddr': 0xfffc, 'duration': 60, 'tcsignificance': 0}
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x05%6\x0f\xfc\xff<\x00&'
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.ZDO command_id=54 data=b'\x00' length=1 fcs=82>
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP ZDO mgmtPermitJoinReq tsn: None {'status': 0}
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=203 data=b'<' length=1 fcs=179>
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO permitJoinInd tsn: None {'duration': 60}
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.api] --> SREQ ZDO mgmtPermitJoinReq tsn: None {'addrmode': 15, 'dstaddr': 65532, 'duration': 60, 'tcsignificance': 0}
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x05%6\x0f\xfc\xff<\x00&'
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=182 data=b'\x00\x00\x00' length=3 fcs=240>
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO mgmtPermitJoinRsp tsn: None {'srcaddr': 0x0000, 'status': 0}
2020-05-02 11:44:17 WARNING (MainThread) [zigpy_cc.zigbee.application] missing tsn from mgmtPermitJoinRsp, maybe not a reply
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.ZDO command_id=54 data=b'\x00' length=1 fcs=82>
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP ZDO mgmtPermitJoinReq tsn: None {'status': 0}
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=182 data=b'\x00\x00\x00' length=3 fcs=240>
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO mgmtPermitJoinRsp tsn: None {'srcaddr': 0x0000, 'status': 0}
2020-05-02 11:44:17 WARNING (MainThread) [zigpy_cc.zigbee.application] missing tsn from mgmtPermitJoinRsp, maybe not a reply
2020-05-02 11:44:22 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=196 data=b'w\x15\x00' length=3 fcs=224>
2020-05-02 11:44:22 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO srcRtgInd tsn: None {'dstaddr': 0x1577, 'relaycount': 0, 'relaylist': []}
2020-05-02 11:44:22 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: srcRtgInd
2020-05-02 11:44:22 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=196 data=b'w\x15\x00' length=3 fcs=224>
2020-05-02 11:44:22 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO srcRtgInd tsn: None {'dstaddr': 0x1577, 'relaycount': 0, 'relaylist': []}
2020-05-02 11:44:22 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: srcRtgInd
2020-05-02 11:44:24 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=196 data=b'w\x15\x00' length=3 fcs=224>
2020-05-02 11:44:24 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO srcRtgInd tsn: None {'dstaddr': 0x1577, 'relaycount': 0, 'relaylist': []}
2020-05-02 11:44:24 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: srcRtgInd
2020-05-02 11:44:24 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=196 data=b'w\x15\x00' length=3 fcs=224>
2020-05-02 11:44:24 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO srcRtgInd tsn: None {'dstaddr': 0x1577, 'relaycount': 0, 'relaylist': []}
2020-05-02 11:44:24 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: srcRtgInd
2020-05-02 11:44:36 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=196 data=b'w\x15\x00' length=3 fcs=224>
2020-05-02 11:44:36 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO srcRtgInd tsn: None {'dstaddr': 0x1577, 'relaycount': 0, 'relaylist': []}
2020-05-02 11:44:36 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: srcRtgInd
2020-05-02 11:44:36 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=196 data=b'w\x15\x00' length=3 fcs=224>
2020-05-02 11:44:36 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO srcRtgInd tsn: None {'dstaddr': 0x1577, 'relaycount': 0, 'relaylist': []}
2020-05-02 11:44:36 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: srcRtgInd
2020-05-02 11:44:38 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=196 data=b'w\x15\x00' length=3 fcs=224>
[... cut ...]
2020-05-02 11:44:42 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP ZDO mgmtPermitJoinReq tsn: None {'status': 0}
2020-05-02 11:44:42 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=182 data=b'\x00\x00\x00' length=3 fcs=240>
2020-05-02 11:44:42 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO mgmtPermitJoinRsp tsn: None {'srcaddr': 0x0000, 'status': 0}
2020-05-02 11:44:42 WARNING (MainThread) [zigpy_cc.zigbee.application] missing tsn from mgmtPermitJoinRsp, maybe not a reply
````

Adminiuga, 2020-05-03:
@sanyatuning 

One thing I'm noticing for ZDO broadcast we set the "addr mode" to `0x0F` 
```
2020-05-02 11:44:17 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x05%6\x0f\xfc\xff<\x00&'
```
https://github.com/zigpy/zigpy-cc/blob/649cf944da340192f346b90edca0da82099310d7/zigpy_cc/zigbee/application.py#L269

Per sniffer it does look like the broadcast goes out, but should the addr mode be `0xFF` for broadcast? In Test API pdf revision 1.14 for broadcast it looks like it should be 0xFF
[image]

In the next command they expand available dst addr modes
[image]

puddly, 2020-05-03:
@Adminiuga 

The spec PDF has a ton of mistakes. The firmware itself uses `0x0F` for the broadcast address mode:

```C
enum
{
  AddrNotPresent = 0,
  AddrGroup = 1,
  Addr16Bit = 2,
  Addr64Bit = 3,
  AddrBroadcast = 15
};
```

Adminiuga, 2020-05-03:
So, it's documented in form of source code then :)

sanyatuning, 2020-05-03:
@Adminiuga 
Thank you for the suggestion, but with 0xff it responds with error status
0x0F is comes form [zigbee-herdsman](https://github.com/Koenkk/zigbee-herdsman/blob/c09dc33db78ebe650b2272d707c958a9ce43d2c8/src/adapter/z-stack/adapter/zStackAdapter.ts#L160)

sanyatuning, 2020-05-03:
@stevetz 
The log looks good to me.
Try unplug the USB stick
If you have another zigbee coordinators turn them off while you pairing.

stevetz, 2020-05-07:
Thanks for the suggestion, and sorry for the delay getting back - I've now sorted this... by switching to zigbee2mqtt. However, this may be relevant: I also couldn't get z2m to work! The logs showed that 'unknown' routers were trying to connect, each device was unknown... I unbound the cc2531, rebooted the VM, re bound the usb stick again and it worked fine! So possibly that is the 'fix' for ZHA as well.... at least in my case - however I'm not able now to fiddle with it any more. I already lost a ton of WAF points doing this!!
Thanks

closed 2020-05-07 by Adminiuga

roger-, 2020-05-15:
FWIW I had a similar issue where I couldn't pair any new devices with ZHA and would only see the same `missing tsn` messages (even though paired devices were apparently still working). In my case the fix was to reconnect the CC2531.

Recently I had another issue where the CC2531 would constant disconnect and the solution there was to reflash the firmware. So there's definitely some stability issues related to hardware/firmware.