zigpy/zigpy-cc #45: ZHA on HA: messages received but not being parsed?

roger-, 2020-05-03:
I have several Aqara sensors that used to work, but suddenly they all show up as "unknown" device/model despite the last seem time being current. I'm seeing the following error message in the HA log:

```
zigpy_cc.exception.TODO: ('write %s', <ParameterType.LIST_NEIGHBOR_LQI: 15>)
```

Along with:

```
 Unhandled message: Subsystem.ZDO concentratorIndCb
```

Trying to re-pair the device doesn't change anything (still shows up as "unknown").

Any ideas?

sanyatuning, 2020-05-03:
fixed, release soon

roger-, 2020-05-05:
I upgraded to 0.4.2 and got this:

```
AttributeError: module 'zigpy.application' has no attribute 'CONFIG_SCHEMA'
```

So I upgraded bellows to 0.16.1 and now I get this:

```
Error setting up entry /dev/ttyACM0 for zha
Traceback (most recent call last):
  File "/home/brent/Software/homeassistant/lib/python3.8/site-packages/homeassistant/config_entries.py", line 215, in async_setup
    result = await component.async_setup_entry(  # type: ignore
  File "/home/brent/Software/homeassistant/lib/python3.8/site-packages/homeassistant/components/zha/__init__.py", line 108, in async_setup_entry
    await zha_gateway.async_initialize()
  File "/home/brent/Software/homeassistant/lib/python3.8/site-packages/homeassistant/components/zha/core/gateway.py", line 146, in async_initialize
    self.application_controller = radio_details[CONTROLLER](radio, database)
TypeError: Can't instantiate abstract class ControllerApplication with abstract methods probe
```

Not sure if it's a HA issue at this point?

sanyatuning, 2020-05-05:
Pls try 0.3.2
0.4 need newer zigpy what is not HA ready as I know

Adminiuga, 2020-05-05:
gonna be in HA dev once https://github.com/home-assistant/core/pull/35161 goes through.

roger-, 2020-05-05:
Thanks.

Giving it a try but now getting a bunch of kernel errors:

```
cdc_acm 1-1.5:1.0: failed to set dtr/rts
```

and re-connections from my TI-CC.

Might be a hardware issue.

roger-, 2020-05-05:
Had to reflash the TI-CC firmware to get it working. Now with 0.3.2 I don't get the same error message but I still see:

```
missing tsn from activeEpRsp, maybe not a reply
missing tsn from mgmtPermitJoinRsp, maybe not a reply
```

Trying to re-pair an Aqara sensor still has it show up as "unknown" and I get the following log:

```
Device 0x482d (00:15:8d:00:04:23:d0:65) joined the network
Device 00:15:8d:00:04:23:d0:65 changed id (0x5b52 => 0x482d)
[0x482d:zdo] ZDO request ZDOCmd.Device_annce: [0x482d, 00:15:8d:00:04:23:d0:65, 128]
Checking quirks for None None (00:15:8d:00:04:23:d0:65)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} set()
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} set()
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} set()
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} set()
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} set()
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} set()
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} set()
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} set()
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} set()
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} set()
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} set()
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} set()
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} set()
[... 17 lines cut ...]
Fail because endpoint list mismatch: {232, 230} set()
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} set()
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because endpoint list mismatch: {1} set()
device - 0x482d:00:15:8d:00:04:23:d0:65 entering async_device_initialized - is_new_join: False
device - 0x482d:00:15:8d:00:04:23:d0:65 has been reset and re-added or its nwk address changed
skipping discovery for previously discovered device - 0x482d:00:15:8d:00:04:23:d0:65
[0x482d](unk_model): started configuration
[0x482d:ZDO](unk_model): 'async_configure' stage succeeded
[0x482d](unk_model): completed configuration
[0x482d](unk_model): stored in registry: ZhaDeviceEntry(name='LUMI lumi.weather', ieee='00:15:8d:00:04:23:d0:65', last_seen=1588718766.270704)
Ignoring message (b'18000a0500420c6c756d692e7765617468657201002003') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18010a01ff42250121a90b0421a80105210b000624010000000064298f0865219a11662b168301000a210000') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18020a0000299608') on cluster 1026: unknown endpoint or cluster id: 1
Ignoring message (b'18030a0000211812') on cluster 1029: unknown endpoint or cluster id: 1
Ignoring message (b'18040a000029df03140028ff100029b626') on cluster 1027: unknown endpoint or cluster id: 1
Ignoring message (b'18050a0000299308') on cluster 1026: unknown endpoint or cluster id: 1
Ignoring message (b'18060a0000213411') on cluster 1029: unknown endpoint or cluster id: 1
Ignoring message (b'18070a000029de03140028ff100029b526') on cluster 1027: unknown endpoint or cluster id: 1
Ignoring message (b'18080a0000299a08') on cluster 1026: unknown endpoint or cluster id: 1
Ignoring message (b'18090a0000213810') on cluster 1029: unknown endpoint or cluster id: 1
Ignoring message (b'180a0a000029de03140028ff100029b526') on cluster 1027: unknown endpoint or cluster id: 1
Ignoring message (b'180b0a0500420c6c756d692e77656174686572') on cluster 0: unknown endpoint or cluster id: 1
```

Adminiuga, 2020-05-05:
It looks like the original discovery of the device was not completed. Remove the device from ZHA (config panel -> ZHA -> Pick the device -> Remove device. Wait a minute, restart HA and confirm the device is not shown in ZHA config panel anymore. After that reset and re-pair the sensor and while pairing keep pressing a button on the device like every 1.5s to keep the device awake

roger-, 2020-05-06:
Thanks, I removed and re-paired a few devices and some of the others just started working.

Ultimately the fix was to update zigpy-cc to 0.3.2.

closed 2020-05-06 by roger-