zigpy/zigpy-cc #54: OTA fails with Tradfri outlet

roger-, 2020-06-11:
I'm using HA 0.111.0 with a CC2531 dongle and trying to update an IKEA Tradfri outlet (firmware 2.0.019) following [this guide](https://www.reddit.com/r/homeassistant/comments/fak430/how_to_update_your_ikea_or_ledevance_firmware/) but I get the following when I issue the `image_notify` command:

```
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy.zcl] [0x3040:1:0x0019] Schema and args lengths do not match in reply
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.zigbee.application] request (0x3040, 260, 25, 1, 1, 79, b'\x19O\x00', False, False)
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.api] --> SREQ AF dataRequest tsn: 79 {'dstaddr': 12352, 'destendpoint': 1, 'srcendpoint': 1, 'clusterid': 25, 'transid': 80, 'options': 0, 'radius': 30, 'len': 3, 'data': b'\x19O\x00'}
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\r$\x01@0\x01\x01\x19\x00P\x00\x1e\x03\x19O\x00Z'
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.AF command_id=1 data=b'\x00' length=1 fcs=100>
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP AF dataRequest tsn: None {'status': 0}
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.AF command_id=128 data=b'\x00\x01P' length=3 fcs=150>
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF dataConfirm tsn: None {'status': 0, 'endpoint': 1, 'transid': 80}
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: dataConfirm
```

Nothing else seems to happen.

Same issue was reported [here](https://community.home-assistant.io/t/ikea-bulb-firmware-upgrade-fails/185578) with an IKEA bulb.

Adminiuga, 2020-06-11:
@sanyatuning can you take a look? We think https://github.com/zigpy/zigpy-cc/blob/23e5cc937fd95dbf276c09b43a7cc8b6750a59e0/zigpy_cc/zigbee/start_znp.py#L55 should be the "in" cluster?

sanyatuning, 2020-06-12:
You are right, thanks for the debug.
I did not test it, but it was definitely wrong.

roger-, 2020-06-26:
I manually added this patch, rebooted and tried `image_notify` but again nothing seemed to happen. 

But it seems like the update  happened in the background at some point in the last two weeks because now my outlet is reporting firmware 2.0.024.

Seems like this is fixed.

closed 2020-06-26 by roger-