zigpy/zigpy-cc #55: Aqara relay. Only one channel works.

3AXAPOB, 2020-06-23:
Hi. I am using cc2531 and have some problems with two channel aqara relay (LUMI lumi.relay.c2acn01). I can switch from frontend only first relay. If I press second channel switch there is no result and got this at log:

2020-06-23 22:26:25 DEBUG (MainThread) [homeassistant.components.zha.entity] light.lumi_lumi_relay_c2acn01_5bd84b04_on_off_2: turned off: 
2020-06-23 22:26:25 WARNING (MainThread) [zigpy_cc.api] Waiter timeout: <Waiter matcher=<Matcher type=CommandType.AREQ subsystem=Subsystem.AF command=dataConfirm payload={'endpoint': 2, 'transid': 102}> future=<Future cancelled> timeout=10000 sequence=None>
2020-06-23 22:26:27 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xfaa2:2:0x0006]: command failed: on exception: 
2020-06-23 22:26:27 DEBUG (MainThread) [homeassistant.components.zha.entity] light.lumi_lumi_relay_c2acn01_5bd84b04_on_off_2: turned on: {'on_off': TimeoutError()}

sanyatuning, 2020-06-27:
I think you need a https://github.com/zigpy/zha-device-handlers for this device
I have some Aqara wall switch, and I had the same problem.
I fixed my issue with these handlers:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/ctrl_ln.py
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/ctrl_neutral.py  

Without the device it is hard to fix your issue.
Can you try to fix by yourself?

3AXAPOB, 2020-06-27:
I will try. But i need some instructions how to add handler to my setup.

Hedda, 2020-06-29:
@3AXAPOB FYI; Home Assistant's ZHA integration component already uses the "ZHA Device Handlers" library by default for ZHA exception and deviation handling

https://www.home-assistant.io/integrations/zha/

Check out:

https://github.com/zigpy/zha-device-handlers

So what you need to do is create a new quirk or update the existing quirks, or submit an issue there with more detailed information if you need help with the creation or updating quirks in the "ZHA Device Handlers" library:

https://github.com/zigpy/zha-device-handlers/blob/dev/CONTRIBUTING.md

PS: There are also instructions in their readme file now on how to test new releases that are not yet available in Home Assistant:

https://github.com/zigpy/zha-device-handlers/blob/dev/README.md