zigpy/zigpy-cc #57: Hue motion sensor not work

mateuszlukasik-PL, 2020-07-04:
**Problem**

When I add Hue motion sensor (SML001) thru ZHA integration, pairing process ends, but none of the sensor function work.
Illuminance, temperature, power is "unknown" state, occupancy is always "clear".

I noticed something strange during pairing operation. When I start adding new devices, after few minutes searching stops, logs stops, but when I only hit "search again", then process is continuing, new entry logs came's up.

Reset and reconfig doesn't help. Tried at HA 111.0, 111.4 and 112.0 version.

Adminiuga sugested this could be a zigpy-cc problem.
[https://github.com/home-assistant/core/issues/36162](url)

**Environment**
[image]
Zigbee module : CC2531 USB

**ZHA log**
Device 0xcf1a (00:17:88:01:06:f7:e9:9d) joined the network
[0xcf1a:zdo] ZDO request ZDOCmd.Device_annce: [0xcf1a, 00:17:88:01:06:f7:e9:9d, 128]
[0xcf1a] Requesting 'Node Descriptor'
Tries remaining: 2
[0xcf1a] Extending timeout for 0x03 request
[0xcf1a] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4107 maximum_buffer_size=89 maximum_incoming_transfer_size=63 server_mask=0 maximum_outgoing_transfer_size=63 descriptor_capability_field=0>
[0xcf1a] Discovering endpoints
Tries remaining: 3
[0xcf1a] Extending timeout for 0x05 request
[0xcf1a] Discovered endpoints: [2, 1]
[0xcf1a:2] Discovering endpoint information
Tries remaining: 3
[0xcf1a] Extending timeout for 0x07 request
[0xcf1a:2] Discovered endpoint information: <Optional endpoint=2 profile=260 device_type=263 device_version=0 input_clusters=[0, 1, 3, 1030, 1024, 1026] output_clusters=[25]>
[0xcf1a] Extending timeout for 0x09 request
[0xcf1a] Extending timeout for 0x0b request
[0xcf1a] Extending timeout for 0x0d request
[0xcf1a:2] Manufacturer: None
[0xcf1a:2] Model: None
[0xcf1a:1] Discovering endpoint information
Tries remaining: 3
[0xcf1a] Extending timeout for 0x0f request
[0xcf1a:1] Discovered endpoint information: <Optional endpoint=1 profile=49246 device_type=2128 device_version=2 input_clusters=[0] output_clusters=[0, 3, 4, 6, 8, 768, 5]>
[0xcf1a] Extending timeout for 0x11 request
[0xcf1a:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=17 command_id=Command.Read_Attributes_rsp>
[0xcf1a:1] Manufacturer: Philips
[0xcf1a:1] Model: SML001
Checking quirks for Philips SML001 (00:17:88:01:06:f7:e9:9d)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 2}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1, 2}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
[... 465 lines cut ...]
2020-07-03 20:42:59 DEBUG (MainThread) [zigpy_cc.api] --> SREQ AF dataRequest tsn: 77 {'dstaddr': 53018, 'destendpoint': 2, 'srcendpoint': 2, 'clusterid': 0, 'transid': 78, 'options': 0, 'radius': 30, 'len': 5, 'data': b'\x00M\x00\x07\x00'}
2020-07-03 20:42:59 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x0f$\x01\x1a\xcf\x02\x02\x00\x00N\x00\x1e\x05\x00M\x00\x07\x00\xe0'
2020-07-03 20:42:59 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.AF command_id=1 data=b'\x00' length=1 fcs=100>
2020-07-03 20:42:59 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP AF dataRequest tsn: None {'status': 0}
2020-07-03 20:42:59 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.AF command_id=128 data=b'\x00\x02N' length=3 fcs=139>
2020-07-03 20:42:59 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF dataConfirm tsn: None {'status': 0, 'endpoint': 2, 'transid': 78}
2020-07-03 20:42:59 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: dataConfirm
2020-07-03 20:43:26 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xcf1a:2:0x0001]: failed to get attributes '['battery_voltage', 'battery_percentage_remaining']' on 'power' cluster: 
2020-07-03 20:43:26 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xcf1a:2:0x0019]: initializing channel: from_cache: False
2020-07-03 20:43:27 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xcf1a:2:0x0406]: failed to get attributes '['occupancy']' on 'occupancy' cluster: 
2020-07-03 20:43:27 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xcf1a:2:0x0000]: initializing channel: from_cache: False
2020-07-03 20:43:27 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xcf1a:2:0x0001]: 'async_initialize' stage succeeded
2020-07-03 20:43:27 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xcf1a:2:0x0402]: 'async_initialize' stage succeeded
2020-07-03 20:43:27 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xcf1a:2:0x0400]: 'async_initialize' stage succeeded
2020-07-03 20:43:27 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xcf1a:2:0x0406]: 'async_initialize' stage succeeded
2020-07-03 20:43:27 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xcf1a:2:0x0000]: 'async_initialize' stage succeeded
2020-07-03 20:43:27 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xcf1a:2:0x0019]: 'async_initialize' stage succeeded
2020-07-03 20:43:27 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xcf1a](SML001): power source: Battery or Unknown
2020-07-03 20:43:27 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xcf1a](SML001): completed initialization
2020-07-03 20:43:57 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=203 data=b'\x00' length=1 fcs=143>
2020-07-03 20:43:57 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO permitJoinInd tsn: None {'duration': 0}
2020-07-03 20:43:57 DEBUG (MainThread) [zigpy_cc.api] --> SREQ UTIL ledControl tsn: None {'ledid': 3, 'mode': <LedMode.Off: 0>}
2020-07-03 20:43:57 DEBUG (MainThread) [zigpy_cc.uart] Send: b"\xfe\x02'\n\x03\x00,"
2020-07-03 20:43:57 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.UTIL command_id=10 data=b'\x00' length=1 fcs=108>
2020-07-03 20:43:57 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP UTIL ledControl tsn: None {'status': 0}

lociii, 2020-07-05:
Exact same behavior here with two SML002 on CC2562R1 and CC2531.
Tested with HA 0.111 and 0.112 series.

Adminiuga, 2020-07-07:
@lociii can you post debug logs of motion sensor pairing? See ZHA integration docs on how to enable debug logging

lociii, 2020-07-07:
@Adminiuga sure. Here you go.
As I have no idea what you're looking for this is the full log from start of system until the end of the initialization on my test system. Hope there's not too much noise from devices on my main zigbee network.

HassOS 4.11 (228) 0.112.3 on Raspberry Pi 4 with CC2531

`
2020-07-07 22:55:34 DEBUG (MainThread) [zigpy.appdb] Loading application state from /config/zigbee.db
2020-07-07 22:55:34 INFO (MainThread) [zigpy_cc.zigbee.application] Starting zigpy-cc version: 0.4.4
2020-07-07 22:55:34 DEBUG (MainThread) [zigpy_cc.uart] Connecting on port /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018ED16C3-if00 with boudrate 115200
2020-07-07 22:55:34 DEBUG (MainThread) [zigpy_cc.uart] Connection made
2020-07-07 22:55:35 DEBUG (MainThread) [zigpy_cc.api] --> SREQ SYS version tsn: None {}
2020-07-07 22:55:35 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x00!\x02#'
2020-07-07 22:55:36 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.SYS command_id=128 data=b'\x00\x02\x00\x02\x06\x03' length=6 fcs=194>
2020-07-07 22:55:36 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ SYS resetInd tsn: None {'reason': 0, 'transportrev': 2, 'productid': 0, 'majorrel': 2, 'minorrel': 6, 'hwrev': 3}
2020-07-07 22:55:36 WARNING (MainThread) [zigpy_cc.zigbee.application] Unhandled message: CommandType.AREQ Subsystem.SYS resetInd
2020-07-07 22:55:36 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.SYS command_id=2 data=b'\x02\x00\x02\x06\x03\x90\x154\x01\x02\x00\x00\x00\x00' length=14 fcs=218>
2020-07-07 22:55:36 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP SYS version tsn: None {'transportrev': 2, 'product': 0, 'majorrel': 2, 'minorrel': 6, 'maintrel': 3, 'revision': 20190608}
2020-07-07 22:55:36 DEBUG (MainThread) [zigpy_cc.api] Version response: {'transportrev': 2, 'product': 0, 'majorrel': 2, 'minorrel': 6, 'maintrel': 3, 'revision': 20190608}
2020-07-07 22:55:36 INFO (MainThread) [zigpy_cc.zigbee.application] Detected znp version 'zStack12' ({'transportrev': 2, 'product': 0, 'majorrel': 2, 'minorrel': 6, 'maintrel': 3, 'revision': 20190608})
2020-07-07 22:55:36 INFO (MainThread) [zigpy_cc.zigbee.application] Forming network
2020-07-07 22:55:36 DEBUG (MainThread) [zigpy_cc.api] --> SREQ SYS osalNvRead tsn: None {'id': <NvItemsIds.ZNP_HAS_CONFIGURED_ZSTACK1: 3840>, 'offset': 0, 'len': 1, 'value': b'U'}
2020-07-07 22:55:36 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x03!\x08\x00\x0f\x00%'
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.SYS command_id=8 data=b'\x00\x01U' length=3 fcs=62>
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP SYS osalNvRead tsn: None {'status': 0, 'len': 1, 'value': b'U'}
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.zigbee.start_znp] Item 'hasConfigured' is valid
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.api] --> SREQ SYS osalNvRead tsn: None {'id': <NvItemsIds.ZNP_HAS_CONFIGURED_ZSTACK1: 3840>, 'offset': 0, 'len': 1, 'value': b'U'}
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x03!\x08\x00\x0f\x00%'
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.SYS command_id=8 data=b'\x00\x01U' length=3 fcs=62>
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP SYS osalNvRead tsn: None {'status': 0, 'len': 1, 'value': b'U'}
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.zigbee.start_znp] Item 'hasConfigured' is valid
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.api] --> SREQ SYS osalNvRead tsn: None {'id': <NvItemsIds.CHANLIST: 132>, 'len': 4, 'offset': 0, 'value': b'\x00\x08\x00\x00'}
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x03!\x08\x84\x00\x00\xae'
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.SYS command_id=8 data=b'\x00\x04\x00\x08\x00\x00' length=6 fcs=99>
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP SYS osalNvRead tsn: None {'status': 0, 'len': 4, 'value': b'\x00\x08\x00\x00'}
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.zigbee.start_znp] Item 'channelList' is valid
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.api] --> SREQ SYS osalNvRead tsn: None {'id': <NvItemsIds.PRECFGKEYS_ENABLE: 99>, 'len': 1, 'offset': 0, 'value': b'\x00'}
2020-07-07 22:55:37 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x03!\x08c\x00\x00I'
2020-07-07 22:55:38 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.SYS command_id=8 data=b'\x00\x01\x00' length=3 fcs=107>
2020-07-07 22:55:38 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP SYS osalNvRead tsn: None {'status': 0, 'len': 1, 'value': b'\x00'}
2020-07-07 22:55:38 DEBUG (MainThread) [zigpy_cc.zigbee.start_znp] Item 'networkKeyDistribute' is valid
2020-07-07 22:55:38 DEBUG (MainThread) [zigpy_cc.api] --> SREQ SAPI readConfiguration tsn: None {'id': <NvItemsIds.PRECFGKEY: 98>, 'configid': <NvItemsIds.PRECFGKEY: 98>, 'len': 16, 'offset': 0, 'value': b'\x01\x03\x05\x07\t\x0b\r\x0f\x00\x02\x04\x06\x08\n\x0c\r'}
2020-07-07 22:55:38 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x01&\x04bA'
[... cut ...] x00\x00\x07\x18\x14\n\x1c\x000\x00\xfe\xa5\x1b' length=27 fcs=2>
2020-07-07 23:02:20 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF incomingMsg tsn: None {'groupid': 0, 'clusterid': 513, 'srcaddr': 0x8d10, 'srcendpoint': 1, 'dstendpoint': 1, 'wasbroadcast': 0, 'linkquality': 0, 'securityuse': 0, 'timestamp': 1269089, 'transseqnumber': 0, 'len': 7, 'data': b'\x18\x14\n\x1c\x000\x00'}
2020-07-07 23:02:20 WARNING (MainThread) [zigpy_cc.zigbee.application] Received frame from unknown device: 0x8d10
`

Adminiuga, 2020-07-07:
so it seems to be exactly same behavior. Looks like it does communicate with the device during the initial interview, but then something happens and in won't talk any longer so you get
```
2020-07-07 23:02:10 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x0f75:2:0x0402]: failed to get attributes '['measured_value']' on 'temperature' cluster: 
2020-07-07 23:02:10 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x0f75:2:0x0019]: initializing channel: from_cache: False
2020-07-07 23:02:10 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x0f75:2:0x0001]: failed to get attributes '['battery_voltage', 'battery_percentage_remaining']' on 'power' cluster: 
```

In all this cases a common denominator is CC2531, so I suspect something in CC2531 is the culprit.

lociii, 2020-07-07:
Guess you can generalize that conclusion to all CC* as my main CC2652R1 has the exact same behavior.

Adminiuga, 2020-07-07:
Nah, sorry, I meant to say something about `zigpy-cc` library implementation is the culprit.

sanyatuning, 2020-07-08:
@Adminiuga
I saw some warnings about the concentratorIndCb message.
Can this be relevant?

Adminiuga, 2020-07-08:
I frankly don't know. Need to check the docs what it does exactly, I assume has something to do with routing

lociii, 2020-07-09:
If one of you guys need an SML001 to reproduce/fix, let me know and I'll sponsor it.

sanyatuning, 2020-07-11:
maybe this should be relevant https://github.com/Koenkk/zigbee2mqtt/issues/310

sanyatuning, 2020-07-11:
I also found this site
https://zigbee.blakadder.com/Philips_SML001.html
That said it is not ZHA compatible yet

sanyatuning, 2020-07-11:
Also this https://github.com/Koenkk/zigbee-herdsman-converters/blob/4810ed5c9a318b5bbf4d1a7390ee3096a74b9bc8/devices.js#L2351

Adminiuga, 2020-07-11:
Hrm, that seems to be outdated. I think threre are ZHA users with HUE motion sensors, because one of them was complaining about LED going off on motion.

lociii, 2020-07-12:
> I also found this site
> https://zigbee.blakadder.com/Philips_SML001.html
> That said it is not ZHA compatible yet

https://zigbee.blakadder.com/Philips_SML002.html
Also says that the outdoor variant is...
However the behavior is exactly the same for both as you can see in the logs posted above.

puddly, 2020-07-12:
This is interesting:

```
2020-07-07 22:59:13 DEBUG (MainThread) [zigpy_cc.api] --> SREQ AF dataRequest tsn: 27 {'dstaddr': 3957, 'destendpoint': 1, 'srcendpoint': 1, 'clusterid': 0, 'transid': 28, 'options': 0, 'radius': 30, 'len': 5, 'data': b'\x00\x1b\x00\x07\x00'}
2020-07-07 22:59:13 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x0f$\x01u\x0f\x01\x01\x00\x00\x1c\x00\x1e\x05\x00\x1b\x00\x07\x00K'
2020-07-07 22:59:13 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.AF command_id=1 data=b'\x00' length=1 fcs=100>
2020-07-07 22:59:13 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP AF dataRequest tsn: None {'status': 0}
2020-07-07 22:59:13 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.AF command_id=128 data=b'\xf0\x01\x1c' length=3 fcs=42>
2020-07-07 22:59:13 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF dataConfirm tsn: None {'status': 240, 'endpoint': 1, 'transid': 28}
2020-07-07 22:59:13 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: dataConfirm
```

A status of 240 (`MAC_TRANSACTION_EXPIRED`) is returned for the ZDO request, though I'm unable to figure out what the ZDO request actually is.

Adminiuga, 2020-07-12:
But why zdo if endpoint is 1?

puddly, 2020-07-12:
Oops, you're right. Ignore that part.

One other possibility is that since the source and destination endpoints are always the same (I haven't dug into where in zigpy this is set), you will run into problems when endpoints other than `1` are used due to the profiles being set on application startup:

https://github.com/zigpy/zigpy-cc/blob/e522adff00ea5231616ad1fd9f585fc6b94b7b98/zigpy_cc/zigbee/start_znp.py#L34-L40

Messages are sent to endpoint `2`, which will be sent with the profile set to `0x0101` (Industrial plant monitoring?), not Home Automation (`0x0104`). I had a similar problem with a Bosch motion sensor not liking that profile on endpoint 2.

Try changing `0x0101` to `0x0104` in `zigpy_cc/zigbee/start_znp.py` and see if that helps.

Adminiuga, 2020-07-12:
I think you are on the right track. The src endpoint is set in zigpy.endpoint. Endpoint but really radio should be picking one up. Zigpy-deconz had similar issues and currently defaults to src endpoint 1

sanyatuning, 2020-07-13:
My Xiaomi wall switches also need ep 2.
I created quirks for them.
Maybe this should be fixed in zigpy core.
What do you think?

mateuszlukasik-PL, 2020-07-13:
I've change this line :
"Endpoint(endpoint=2, appprofid=0x0101),"
in homeassistant/lib/zigpy-cc/zibgee/start_znp.py to :
"Endpoint(endpoint=2, appprofid=0x0104),"
saved, reboot, add new zigbee device but result is the same.

**Log :**
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.zigbee.application] broadcast (0, 54, 0, 0, 0, 0, 1, b'\x01<\x00', <BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>)
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.api] --> SREQ ZDO mgmtPermitJoinReq tsn: 1 {'addrmode': 15, 'dstaddr': 0xfffc, 'duration': 60, 'tcsignificance': 0}
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x05%6\x0f\xfc\xff<\x00&'
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.ZDO command_id=54 data=b'\x00' length=1 fcs=82>
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP ZDO mgmtPermitJoinReq tsn: None {'status': 0}
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.api] --> SREQ ZDO mgmtPermitJoinReq tsn: None {'addrmode': 15, 'dstaddr': <BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>, 'duration': 60, 'tcsignificance': 0}
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x05%6\x0f\xfc\xff<\x00&'
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=203 data=b'<' length=1 fcs=179>
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO permitJoinInd tsn: None {'duration': 60}
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.api] --> SREQ UTIL ledControl tsn: None {'ledid': 3, 'mode': <LedMode.On: 1>}
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.uart] Send: b"\xfe\x02'\n\x03\x01-"
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=182 data=b'\x00\x00\x00' length=3 fcs=240>
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO mgmtPermitJoinRsp tsn: None {'srcaddr': 0x0000, 'status': 0}
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.api] --> SREQ UTIL ledControl tsn: None {'ledid': 3, 'mode': <LedMode.On: 1>}
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.uart] Send: b"\xfe\x02'\n\x03\x01-"
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.ZDO command_id=54 data=b'\x00' length=1 fcs=82>
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP ZDO mgmtPermitJoinReq tsn: None {'status': 0}
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.UTIL command_id=10 data=b'\x00' length=1 fcs=108>
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP UTIL ledControl tsn: None {'status': 0}
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.UTIL command_id=10 data=b'\x00' length=1 fcs=108>
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP UTIL ledControl tsn: None {'status': 0}
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=182 data=b'\x00\x00\x00' length=3 fcs=240>
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO mgmtPermitJoinRsp tsn: None {'srcaddr': 0x0000, 'status': 0}
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.api] --> SREQ UTIL ledControl tsn: None {'ledid': 3, 'mode': <LedMode.On: 1>}
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.uart] Send: b"\xfe\x02'\n\x03\x01-"
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.UTIL command_id=10 data=b'\x00' length=1 fcs=108>
2020-07-12 22:28:57 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP UTIL ledControl tsn: None {'status': 0}
2020-07-12 22:28:58 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=202 data=b'\x1a\xcf\x9d\xe9\xf7\x06\x01\x88\x17\x00\x00\x00' length=12 fcs=77>
2020-07-12 22:28:58 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO tcDeviceInd tsn: None {'nwkaddr': 0xcf1a, 'extaddr': 00:17:88:01:06:f7:e9:9d, 'parentaddr': 0x0000}
2020-07-12 22:28:58 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: tcDeviceInd
2020-07-12 22:28:59 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=193 data=b'\x1a\xcf\x1a\xcf\x9d\xe9\xf7\x06\x01\x88\x17\x00\x80' length=13 fcs=18>
2020-07-12 22:28:59 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO endDeviceAnnceInd tsn: 0 {'srcaddr': 0xcf1a, 'nwkaddr': 0xcf1a, 'ieeeaddr': 00:17:88:01:06:f7:e9:9d, 'capabilities': 128}
2020-07-12 22:28:59 INFO (MainThread) [zigpy_cc.zigbee.application] New device joined: 0xcf1a, 00:17:88:01:06:f7:e9:9d
2020-07-12 22:28:59 INFO (MainThread) [zigpy.application] Device 0xcf1a (00:17:88:01:06:f7:e9:9d) joined the network
2020-07-12 22:28:59 DEBUG (MainThread) [zigpy_cc.zigbee.application] REPLY for 0 endDeviceAnnceInd
2020-07-12 22:28:59 INFO (MainThread) [zigpy_cc.zigbee.application] handle_message endDeviceAnnceInd
2020-07-12 22:28:59 DEBUG (MainThread) [zigpy.zdo] [0xcf1a:zdo] ZDO request ZDOCmd.Device_annce: [0xcf1a, 00:17:88:01:06:f7:e9:9d, 128]
2020-07-12 22:28:59 INFO (MainThread) [zigpy.device] [0xcf1a] Requesting 'Node Descriptor'
[... cut ...]
2020-07-12 22:32:54 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xcf1a:2:0x0402]: 'async_initialize' stage succeeded
2020-07-12 22:32:54 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xcf1a:2:0x0019]: 'async_initialize' stage succeeded
2020-07-12 22:32:54 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xcf1a](SML001): power source: Battery or Unknown
2020-07-12 22:32:54 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xcf1a](SML001): completed initialization

lociii, 2020-07-24:
https://github.com/zigpy/zha-device-handlers/issues/387

resoai, 2020-07-31:
I have the very same issue with SML002 on CC2538.

jvalst, 2020-08-17:
Same issue here with the sml001 with my sonoff zigbee dongle cc2531. First my motion sensor didn't got any entities only the device, 2nd try it showed me also entities but as mentioned above, they won't work.

jvalst, 2020-08-30:
Any news on this one? Can we help someone with logs or stuff? Should be awesome to have this motion sensor available in ZHA by that dongle.

puddly, 2020-09-25:
@jvalst @lociii I believe this bug has been fixed in https://github.com/zha-ng/zigpy-znp/ and @Adminiuga has tested this sensor specifically with it. Let me know if it works for you, especially if you're using CC2531s running Z-Stack Home 1.2 (install the latest version from Git, support hasn't officially been released yet).

mateuszlukasik-PL, 2020-09-26:
Yes, it works!
Great job! Thanks!

But I've flashed CC2531ZNP-without-SBL.bin (Z-Stack 3.0.1) to my CC2531 stick.

lociii, 2020-09-27:
@puddly I have the feeling this is a bit mixed up now. You're talking about it being fixed in zigpy-znp, however this is a bug report for zigpy-cc.
For me the SML002 is still not working in zigpy-cc on HA 0.115.3 with zigpy-cc 0.5.2.

@mateuszlukasik-PL What are you running? Does the SML001 report values?

pipiche38, 2020-09-27:
I do not know what exactly is the problem , but as far as I known SML001 and SML002 required to bind the various cluster in a particular order.

1/ 0x0406
2/ 0x0400
3/ 0x0402
4/ 0x0001

If you bind them in a different order then most-likely the motion won't work and the lux might not as well

jvalst, 2020-09-27:
Oh, I would really try out this one, but someone has to guide me through updating zigpy-cc. I just bought the cc usb stick (https://nl.aliexpress.com/item/4000859025741.html?spm=a2g0s.9042311.0.0.735c4c4dkGEHiA) and plugged it in into my Rpi.

lociii, 2020-09-27:
@jvalst That's exactly what it looked like for me. Have switched now to zigpy-znp as I'm running a CC2562R1 with ZStack 3
@puddly Can confirm it works properly on zigpy-znp.

jvalst, 2020-09-27:
Yeah as it looked in that link, I'm running znp as well. I'm not really into the whole zigpy stuff, so not sure where to start to upgrade it. I'm running latest home assistant, and never touched the firmware of the usb stick itself.