zigpy/zigpy-cc #58: Zigbee groups don't switch Tradfri light bulbs

lociii, 2020-07-12:
Hey folks,

first of all: thanks for your amazing work on supporting CC* in ZHA.
We, as the community, cannot appreciate the work of all the volunteers enough!

**Setup**

Home Assistant 0.112.3 (running on supervised Ubuntu)
zigpy 0.21.0
zigpy-cc 0.4.4
Controller: LAUNCHXL-CC26X2R1
Bulbs: TRADFRI bulb E14 CWS opal 600lm by IKEA of Sweden

**Problem**

Switching zigbee groups on or off does not change the state of the devices included in the group.

**Device signature of one of the bulbs** (all are the same)

```
{
  "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4476 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 49246,
      "device_type": "0x0200",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x0b05",
        "0x1000"
      ],
      "out_clusters": [
        "0x0005",
        "0x0019",
        "0x0020",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI bulb E14 CWS opal 600lm",
  "class": "zigpy.device.Device"
}
```

**Log**

Turn group on
```
2020-07-12 14:24:03 DEBUG (MainThread) [zigpy_cc.zigbee.application] multicast (0x0002, 260, 6, 1, 175, b'\x01\xaf\x01', 0, 3)
2020-07-12 14:24:03 DEBUG (MainThread) [zigpy_cc.api] --> SREQ AF dataRequest tsn: 175 {'dstaddr': 2, 'destendpoint': 1, 'srcendpoint': 1, 'clusterid': 6, 'transid': 176, 'options': 0, 'radius': 30, 'len': 3, 'data': b'\x01\xaf\x01'}
2020-07-12 14:24:03 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\r$\x01\x02\x00\x01\x01\x06\x00\xb0\x00\x1e\x03\x01\xaf\x01.'
2020-07-12 14:24:03 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.AF command_id=1 data=b'\x00' length=1 fcs=100>
2020-07-12 14:24:03 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP AF dataRequest tsn: None {'status': 0}
2020-07-12 14:24:03 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.AF command_id=128 data=b'\xcd\x01\xb0' length=3 fcs=187>
2020-07-12 14:24:03 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF dataConfirm tsn: None {'status': 205, 'endpoint': 1, 'transid': 176}
2020-07-12 14:24:03 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: dataConfirm
2020-07-12 14:24:03 DEBUG (MainThread) [homeassistant.components.zha.entity] light.stehlampe_gastezimmer_zha: turned on: {'on_off': [1, <Status.SUCCESS: 0>]}
```

Turn group off
```
2020-07-12 14:24:11 DEBUG (MainThread) [zigpy_cc.zigbee.application] multicast (0x0002, 260, 6, 1, 177, b'\x01\xb1\x00', 0, 3)
2020-07-12 14:24:11 DEBUG (MainThread) [zigpy_cc.api] --> SREQ AF dataRequest tsn: 177 {'dstaddr': 2, 'destendpoint': 1, 'srcendpoint': 1, 'clusterid': 6, 'transid': 178, 'options': 0, 'radius': 30, 'len': 3, 'data': b'\x01\xb1\x00'}
2020-07-12 14:24:11 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\r$\x01\x02\x00\x01\x01\x06\x00\xb2\x00\x1e\x03\x01\xb1\x003'
2020-07-12 14:24:11 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.AF command_id=1 data=b'\x00' length=1 fcs=100>
2020-07-12 14:24:11 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP AF dataRequest tsn: None {'status': 0}
2020-07-12 14:24:12 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.AF command_id=128 data=b'\xcd\x01\xb2' length=3 fcs=185>
2020-07-12 14:24:12 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF dataConfirm tsn: None {'status': 205, 'endpoint': 1, 'transid': 178}
2020-07-12 14:24:12 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: dataConfirm
2020-07-12 14:24:12 DEBUG (MainThread) [homeassistant.components.zha.entity] light.stehlampe_gastezimmer_zha: turned off: [0, <Status.SUCCESS: 0>]
```

Thanks so much for having a look at it.

sanyatuning, 2020-07-12:
Do you mean not change the state in HA or do nothing at all?
Can you turn on/off the bulbs one by one?

lociii, 2020-07-12:
Nothing at all. Bulbs don't turn on/off.
They can be fully controlled individually.

saz, 2020-07-12:
Same setup, same problem for me.

lociii, 2020-07-17:
Same behavior for groups of various ledvance bulbs. Seems like a general zha group issue with CC* controllers.

sanyatuning, 2020-07-19:
Can you test with branch "refact"?

lociii, 2020-07-19:
Will get some more bulbs for my test setup tomorrow and report back.

sanyatuning, 2020-07-19:
Probably you need to repair your devices.
The default channel list is now read from zigpy.config object.

lociii, 2020-07-21:
Here's my whole log from boot, adding the controller and trying to pair a tradfri bulb.
https://pastebin.com/UN4w0jAT

HA 0.112.4
HassOS 4.11
CC 2531 with zigpy-cc 0.5.0b0

sanyatuning, 2020-07-23:
@lociii in your log: 
```
2020-07-21 15:08:39 INFO (MainThread) [zigpy_cc.zigbee.application] Starting zigpy-cc version: 0.4.4
```

lociii, 2020-07-23:
@sanyatuning sorry for me being stupid.
Tested it again now in a devcontainer with properly upgraded dependencies.
Group gets initialized properly and lights can be controlled by it. Great job!!!

Will there be any way to prevent repairing when this goes onto production instances?

commit f345d443 (2020-07-19): pull updates from zigbee-herdsman
fix zigbee groups fixes #58
PR #60 (2020-08-13): fix zigbee groups, updates form zigbee-herdsman
fix zigbee groups fixes #58
pull updates from zigbee-herdsman
commit 5625ea5d (2020-08-13): fix zigbee groups, updates form zigbee-herdsman (#60)

* pull updates from zigbee-herdsman
fix zigbee groups fixes #58

* add test

* update broadcast based on zigbee-herdsman

* ignore resetInd, do not log waiting for...
closed 2020-08-13 by sanyatuning