zigpy/zigpy-cc #59: CC2531 + WXKG01LM unknown endpoint

dvv, 2020-07-17:
Lumi round button WXKG01LM seems to pair ok but device is not registered:

```
homeassistant_1  | 2020-07-17 08:43:36 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=201 data=b'\xab&\xf3\xb7\x1c\x03\x00\x8d\x15\x00\x00\x00\x00' length=13 fcs=207>
homeassistant_1  | 2020-07-17 08:43:36 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO leaveInd tsn: None {'srcaddr': 0x26ab, 'extaddr': 00:15:8d:00:03:1c:b7:f3, 'request': 0, 'removechildren': 0, 'rejoin': 0}
homeassistant_1  | 2020-07-17 08:43:36 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: leaveInd
homeassistant_1  | 2020-07-17 08:43:37 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=202 data=b'\xce\xaa\xf3\xb7\x1c\x03\x00\x8d\x15\x00\x00\x00' length=12 fcs=36>
homeassistant_1  | 2020-07-17 08:43:37 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO tcDeviceInd tsn: None {'nwkaddr': 0xaace, 'extaddr': 00:15:8d:00:03:1c:b7:f3, 'parentaddr': 0x0000}
homeassistant_1  | 2020-07-17 08:43:37 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: tcDeviceInd
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=193 data=b'\xce\xaa\xce\xaa\xf3\xb7\x1c\x03\x00\x8d\x15\x00\x80' length=13 fcs=202>
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO endDeviceAnnceInd tsn: 0 {'srcaddr': 0xaace, 'nwkaddr': 0xaace, 'ieeeaddr': 00:15:8d:00:03:1c:b7:f3, 'capabilities': 128}
homeassistant_1  | 2020-07-17 08:43:38 INFO (MainThread) [zigpy_cc.zigbee.application] New device joined: 0xaace, 00:15:8d:00:03:1c:b7:f3
homeassistant_1  | 2020-07-17 08:43:38 INFO (MainThread) [zigpy.application] Device 0xaace (00:15:8d:00:03:1c:b7:f3) joined the network
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy.application] Device 00:15:8d:00:03:1c:b7:f3 changed id (0x26ab => 0xaace)
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.zigbee.application] REPLY for 0 endDeviceAnnceInd
homeassistant_1  | 2020-07-17 08:43:38 INFO (MainThread) [zigpy_cc.zigbee.application] handle_message endDeviceAnnceInd
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy.zdo] [0xaace:zdo] ZDO request ZDOCmd.Device_annce: [0xaace, 00:15:8d:00:03:1c:b7:f3, 128]
homeassistant_1  | 2020-07-17 08:43:38 INFO (MainThread) [zigpy.device] [0xaace] Requesting 'Node Descriptor'
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy.device] [0xaace] Extending timeout for 0x0f request
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.zigbee.application] request (0xaace, 0, <ZDOCmd.Node_Desc_req: 2>, 0, 0, 15, b'\x0f\xce\xaa', True, False)
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.api] waiting for 15 nodeDescReq
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.api] --> SREQ ZDO nodeDescReq tsn: 15 {'dstaddr': 0xaace, 'nwkaddrofinterest': 0xaace}
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x04%\x02\xce\xaa\xce\xaa#'
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.ZDO command_id=2 data=b'\x00' length=1 fcs=102>
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP ZDO nodeDescReq tsn: None {'status': 0}
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.AF command_id=129 data=b'\x00\x00\x00\x00\xce\xaa\x01\x01\x00<\x00<_^\x00\x00\x19\x18\x00\n\x05\x00B\x12lumi.sensor_switch\xce\xaa\x1d' length=45 fcs=194>
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF incomingMsg tsn: None {'groupid': 0, 'clusterid': 0, 'srcaddr': 0xaace, 'srcendpoint': 1, 'dstendpoint': 1, 'wasbroadcast': 0, 'linkquality': 60, 'securityuse': 0, 'timestamp': 6184764, 'transseqnumber': 0, 'len': 25, 'data': b'\x18\x00\n\x05\x00B\x12lumi.sensor_switch'}
homeassistant_1  | 2020-07-17 08:43:38 INFO (MainThread) [zigpy_cc.zigbee.application] handle_message incomingMsg
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy.device] Ignoring message (b'18000a050042126c756d692e73656e736f725f737769746368') on cluster 0: unknown endpoint or cluster id: 1
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.AF command_id=129 data=b'\x00\x00\x00\x00\xce\xaa\x01\x01\x009\x00K_^\x00\x00\x07\x18\x01\n\x01\x00 \n\xce\xaa\x1d' length=27 fcs=143>
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF incomingMsg tsn: None {'groupid': 0, 'clusterid': 0, 'srcaddr': 0xaace, 'srcendpoint': 1, 'dstendpoint': 1, 'wasbroadcast': 0, 'linkquality': 57, 'securityuse': 0, 'timestamp': 6184779, 'transseqnumber': 0, 'len': 7, 'data': b'\x18\x01\n\x01\x00 \n'}
homeassistant_1  | 2020-07-17 08:43:38 INFO (MainThread) [zigpy_cc.zigbee.application] handle_message incomingMsg
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy.device] Ignoring message (b'18010a0100200a') on cluster 0: unknown endpoint or cluster id: 1
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.AF command_id=129 data=b'\x00\x00\x00\x00\xce\xaa\x01\x01\x009\x00W_^\x00\x00\x1d\x1c4\x12\x02\n\x02\xffL\x06\x00\x10\x01!\xce\x0b!\xa8\x01$\x00\x00\x00\x00\x00!\x18\x00 W\xce\xaa\x1d' length=49 fcs=9>
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF incomingMsg tsn: None {'groupid': 0, 'clusterid': 0, 'srcaddr': 0xaace, 'srcendpoint': 1, 'dstendpoint': 1, 'wasbroadcast': 0, 'linkquality': 57, 'securityuse': 0, 'timestamp': 6184791, 'transseqnumber': 0, 'len': 29, 'data': b'\x1c4\x12\x02\n\x02\xffL\x06\x00\x10\x01!\xce\x0b!\xa8\x01$\x00\x00\x00\x00\x00!\x18\x00 W'}
homeassistant_1  | 2020-07-17 08:43:38 INFO (MainThread) [zigpy_cc.zigbee.application] handle_message incomingMsg
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy.device] Ignoring message (b'1c3412020a02ff4c0600100121ce0b21a8012400000000002118002057') on cluster 0: unknown endpoint or cluster id: 1
```

On the other hand, Water Leak sensor SJCGQ11LM registered just fine:

```
homeassistant_1  | 2020-07-17 08:52:45 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=201 data=b'\xff\x19v\x14\x01\x04\x00\x8d\x15\x00\x00\x00\x00' length=13 fcs=152>
homeassistant_1  | 2020-07-17 08:52:45 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO leaveInd tsn: None {'srcaddr': 0x19ff, 'extaddr': 00:15:8d:00:04:01:14:76, 'request': 0, 'removechildren': 0, 'rejoin': 0}
homeassistant_1  | 2020-07-17 08:52:45 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: leaveInd
homeassistant_1  | 2020-07-17 08:52:46 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=202 data=b' \xfbv\x14\x01\x04\x00\x8d\x15\x00\x00\x00' length=12 fcs=167>
homeassistant_1  | 2020-07-17 08:52:46 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO tcDeviceInd tsn: None {'nwkaddr': 0xfb20, 'extaddr': 00:15:8d:00:04:01:14:76, 'parentaddr': 0x0000}
homeassistant_1  | 2020-07-17 08:52:46 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: tcDeviceInd
homeassistant_1  | 2020-07-17 08:52:47 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=193 data=b' \xfb \xfbv\x14\x01\x04\x00\x8d\x15\x00\x80' length=13 fcs=246>
homeassistant_1  | 2020-07-17 08:52:47 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO endDeviceAnnceInd tsn: 0 {'srcaddr': 0xfb20, 'nwkaddr': 0xfb20, 'ieeeaddr': 00:15:8d:00:04:01:14:76, 'capabilities': 128}
homeassistant_1  | 2020-07-17 08:52:47 INFO (MainThread) [zigpy_cc.zigbee.application] New device joined: 0xfb20, 00:15:8d:00:04:01:14:76
homeassistant_1  | 2020-07-17 08:52:47 INFO (MainThread) [zigpy.application] Device 0xfb20 (00:15:8d:00:04:01:14:76) joined the network
[... cut ...] 0\x00 \x18\x00\n\x05\x00B\x15lumi.sensor_wleak.aq1\x01\x00 \x04 \xfb\x1d' length=52 fcs=94>
homeassistant_1  | 2020-07-17 08:52:47 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF incomingMsg tsn: None {'groupid': 0, 'clusterid': 0, 'srcaddr': 0xfb20, 'srcendpoint': 1, 'dstendpoint': 1, 'wasbroadcast': 0, 'linkquality': 42, 'securityuse': 0, 'timestamp': 7900714, 'transseqnumber': 0, 'len': 32, 'data': b'\x18\x00\n\x05\x00B\x15lumi.sensor_wleak.aq1\x01\x00 \x04'}
```

Please point out where to dig?

sanyatuning, 2020-07-23:
I have this button and it is paired.
Try pressing the reset button (1 short press in 1-2 sec. period) while pairing to keep the device awake.

dvv, 2020-07-24:
It _is_ paired. But further comm fails.
Let us look at extracted relevant portions of logs. We can see both devices joined network. Then for the leak sensor they iterate thru quirks (correct), but for the button they don't.

button:
```
homeassistant_1  | 2020-07-17 08:43:38 INFO (MainThread) [zigpy.application] Device 0xaace (00:15:8d:00:03:1c:b7:f3) joined the network
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy.application] Device 00:15:8d:00:03:1c:b7:f3 changed id (0x26ab => 0xaace)
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.zigbee.application] REPLY for 0 endDeviceAnnceInd
homeassistant_1  | 2020-07-17 08:43:38 INFO (MainThread) [zigpy_cc.zigbee.application] handle_message endDeviceAnnceInd
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy.zdo] [0xaace:zdo] ZDO request ZDOCmd.Device_annce: [0xaace, 00:15:8d:00:03:1c:b7:f3, 128]
homeassistant_1  | 2020-07-17 08:43:38 INFO (MainThread) [zigpy.device] [0xaace] Requesting 'Node Descriptor'
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy.device] [0xaace] Extending timeout for 0x0f request
homeassistant_1  | 2020-07-17 08:43:38 DEBUG (MainThread) [zigpy_cc.zigbee.application] request (0xaace, 0, <ZDOCmd.Node_Desc_req: 2>, 0, 0, 15, b'\x0f\xce\xaa', True, False)
```

leak:

```
homeassistant_1  | 2020-07-17 08:52:47 INFO (MainThread) [zigpy.application] Device 0xfb20 (00:15:8d:00:04:01:14:76) joined the network
homeassistant_1  | 2020-07-17 08:52:47 DEBUG (MainThread) [zigpy.application] Device 00:15:8d:00:04:01:14:76 changed id (0x19ff => 0xfb20)
homeassistant_1  | 2020-07-17 08:52:47 DEBUG (MainThread) [zigpy_cc.zigbee.application] REPLY for 0 endDeviceAnnceInd
homeassistant_1  | 2020-07-17 08:52:47 INFO (MainThread) [zigpy_cc.zigbee.application] handle_message endDeviceAnnceInd
homeassistant_1  | 2020-07-17 08:52:47 DEBUG (MainThread) [zigpy.zdo] [0xfb20:zdo] ZDO request ZDOCmd.Device_annce: [0xfb20, 00:15:8d:00:04:01:14:76, 128]
homeassistant_1  | 2020-07-17 08:52:47 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for None None (00:15:8d:00:04:01:14:76)
homeassistant_1  | 2020-07-17 08:52:47 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
```

sanyatuning, 2020-08-04:
Sorry but I think your device is not paired.
```
homeassistant_1  | 2020-07-17 08:43:38 INFO (MainThread) [zigpy.application] Device 0xaace (00:15:8d:00:03:1c:b7:f3) joined the network
```
Join is only the start of pairing
Then we need many requests.
You need to keep the device awake while pairing, it should work.
If not, try reset your zigbee dongle with the reset button.