zigpy/zigpy-cc #61: [Request] a zigpy_cc server example

iftahgi, 2020-08-02:
Hi there, 
I need to access sensors' attributes programatically, not through any home automation platform. 
I wonder whether there is any reliable template for server code using the sigpy_cc module (intialisation, instantiation of coordinator, on-boarding devices and instantiating their classes, reading attributes and sending commands to devices, etc.).

Many many thanks!

sanyatuning, 2020-08-04:
It is not easy I think.
You will need https://github.com/zigpy/zigpy too.
Personally I use https://github.com/zigpy/zigpy-cc/blob/master/test.py to test zigpy-cc without HA (check git history for this file)

You may look into ZHA integration as well for inspiration
https://github.com/home-assistant/core/tree/dev/homeassistant/components/zha

closed 2020-08-04 by sanyatuning

Hedda, 2020-08-11:
@iftahgi and @sanyatuning FYI, a very similar question has now been posted for zigpy here -> https://github.com/zigpy/zigpy/issues/452