zigpy/zigpy-cc #63: [SUGGESTION] Suggest zigpy-cc and ZHA devs team-up with slaesh to test his CC2652RB stick as another reference adapter

Hedda, 2020-08-10:
Suggest zigpy-cc + zigpy-znp and ZHA devs team-up with @slaesh to test his new CC2652RB stick as another reference adapter.

@slaesh I believe that you already donated adapters to the Zigbee2mqtt project so wondering if you would also be willing to donate some of your CC2652RB stick adapters to @sanyatuning & @Adminiuga & @puddly & @dmulcahey for [zigpy/zigpy-cc](https://github.com/zigpy/zigpy-cc) and [https://github.com/zha-ng/zigpy-znp](https://github.com/zha-ng/zigpy-znp) development as used with zigpy in [Home Assistant's ZHA integration component for its native Zigbee support](https://www.home-assistant.io/integrations/zha/)?

FYI; @slaesh has developed his new CC2652RB stick as an open source hardware designed similar to the CC2652R stick by Electrolama/omerk as a another friendly competition option, and as noted, slaesh stick is based on CC2652RB instead of CC2652R. His adapter is since recently listed as a supported reference adapter by Zigbee2MQTT on their project page.

https://github.com/slaesh/cc2652-stick

and

https://slae.sh/projects/cc2652/

Like omerk, slaesh is also based in Europe and is selling the finished product (unflashed) via Tindie 

https://www.tindie.com/products/slaesh/cc2652-zigbee-coordinator-or-openthread-router/

slaesh, 2020-08-11:
thanks @Hedda !
Sure, sounds great. Thanks for your request.

Currently stuck during a business trip.

Maybe write me for more details https://t.me/slaesh

Hedda, 2020-08-11:
> thanks @Hedda !
> Sure, sounds great. Thanks for your request.
> 
> Currently stuck during a business trip.
> 
> Maybe write me for more details https://t.me/slaesh

@slaesh That is very nice of you! Could you maybe post that in https://github.com/zigpy/zigpy-cc/issues/63 and https://github.com/zha-ng/zigpy-znp/issues/28 telling @sanyatuning & @Adminiuga & @puddly & @dmulcahey that they can each e-mail their own postal addresses to you if they would like to get you? 

PS: I don't have their contact information or any personal connection to them myself, I am really just a fan of zigpy and ZHA ;)

Hedda, 2020-09-02:
@slaesh Back from your business trip?