zigpy/zigpy-cc #67: [Device Support Request] Lonsonho 2 Gang wall switch

scaytrase, 2020-08-14:
**Is your feature request related to a problem? Please describe.**
Device almost works good without quirk but HA cannot control second (right) switch at the moment

**Describe the solution you'd like**
Control second switch as the first one. Currently issuing any commands to it results in timout error (stactrace below)

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
Full log
```
Device 0xa474 (bc:33:ac:ff:fe:65:5c:e2) joined the network
[0xa474:zdo] ZDO request ZDOCmd.Device_annce: [0xa474, bc:33:ac:ff:fe:65:5c:e2, 142]
[0xa474] Requesting 'Node Descriptor'
Tries remaining: 2
[0xa474] Extending timeout for 0x15 request
[0xa474] Node Descriptor: <Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=0 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0xa474] Discovering endpoints
Tries remaining: 3
[0xa474] Discovered endpoints: [1, 2]
[0xa474:1] Discovering endpoint information
Tries remaining: 3
[0xa474:1] Discovered endpoint information: <Optional endpoint=1 profile=260 device_type=256 device_version=0 input_clusters=[0, 3, 4, 5, 6] output_clusters=[25]>
[0xa474:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=27 command_id=Command.Read_Attributes_rsp>
[0xa474:1] Manufacturer: _TYZB01_6sadkhcy
[0xa474:1] Model: TS0002
[0xa474:2] Discovering endpoint information
Tries remaining: 3
[0xa474:2] Discovered endpoint information: <Optional endpoint=2 profile=260 device_type=256 device_version=0 input_clusters=[0, 3, 4, 5, 6] output_clusters=[25]>
Checking quirks for _TYZB01_6sadkhcy TS0002 (bc:33:ac:ff:fe:65:5c:e2)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 2}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1, 2}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because endpoint list mismatch: {1} {1, 2}
device - 0xa474:bc:33:ac:ff:fe:65:5c:e2 entering async_device_initialized - is_new_join: True
device - 0xa474:bc:33:ac:ff:fe:65:5c:e2 has joined the ZHA zigbee network
[0xa474](TS0002): started configuration
[0xa474:ZDO](TS0002): 'async_configure' stage succeeded
[0xa474:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0xa474:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0xa474:1:0x0000]: finished channel configuration
[0xa474:1:0x0019]: bound 'ota' cluster: Status.SUCCESS
[0xa474:1:0x0019]: finished channel configuration
[0xa474:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=37 command_id=Command.Default_Response>
[0xa474:1:0x0006]: reporting 'on_off' attr on 'on_off' cluster: 0/900/1: Result: '[6, <Status.UNSUP_GENERAL_COMMAND: 130>]'
[0xa474:1:0x0006]: finished channel configuration
[0xa474:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=39 command_id=Command.Read_Attributes_rsp>
[0xa474:1:0x0000]: initializing channel: from_cache: False
[0xa474:2:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0xa474:1:0x0006]: 'async_configure' stage succeeded
[... 93 lines cut ...]
Logger: homeassistant.components.websocket_api.http.connection.281473257193424
Source: components/zha/core/device.py:549
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 22:04:57 (2 occurrences)
Last logged: 22:08:13

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 130, in handle_call_service
    connection.context(msg),
  File "/usr/src/homeassistant/homeassistant/core.py", line 1260, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1295, in _execute_service
    await handler.func(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 518, in admin_handler
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 925, in issue_zigbee_cluster_command
    manufacturer=manufacturer,
  File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 549, in issue_cluster_command
    command, *args, manufacturer=manufacturer, expect_reply=True
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 210, in request
    result = await asyncio.wait_for(req.result, timeout)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 449, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```

scaytrase, 2020-08-14:
Ok, will try to do this. WIth `debug logs` you mean these https://www.home-assistant.io/integrations/zha/#debug-logging ?

scaytrase, 2020-08-14:
here is the full log. I sanitized some info, but if needed I can share full log privately for reasearch.  On this log I set up zigbee integration, paired my device and do some on-offs on both channels, having left work ok and right giving me timeouts

```
2020-08-14 22:54:34 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event timer_out_of_sync[L]: seconds=1.3030935970000428>
2020-08-14 22:54:40 DEBUG (MainThread) [bellows.ezsp] Resetting EZSP
2020-08-14 22:54:43 DEBUG (MainThread) [bellows.ezsp] Unsuccessful radio probe of '/dev/ttyACM0' port
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/bellows/ezsp.py", line 46, in probe
    await asyncio.wait_for(ezsp._probe(), timeout=PROBE_TIMEOUT)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 449, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
2020-08-14 22:54:43 DEBUG (MainThread) [zigpy_deconz.uart] Connection made
2020-08-14 22:54:43 DEBUG (MainThread) [zigpy_deconz.api] Command Command.device_state (0, 0, 0)
2020-08-14 22:54:43 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x0702000800000000
2020-08-14 22:54:44 WARNING (MainThread) [zigpy_deconz.api] No response to 'Command.device_state' command with seq id '0x02'
2020-08-14 22:54:44 DEBUG (MainThread) [zigpy_deconz.api] Unsuccessful radio probe of '/dev/ttyACM0' port
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigpy_deconz/api.py", line 356, in probe
    await asyncio.wait_for(api._probe(), timeout=PROBE_TIMEOUT)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 442, in wait_for
    return fut.result()
  File "/usr/local/lib/python3.7/site-packages/zigpy_deconz/api.py", line 372, in _probe
    await self.device_state()
  File "/usr/local/lib/python3.7/site-packages/zigpy_deconz/api.py", line 289, in _command
    return await asyncio.wait_for(fut, timeout=COMMAND_TIMEOUT)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 449, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
2020-08-14 22:54:44 DEBUG (MainThread) [zigpy_deconz.uart] Closed serial connection
2020-08-14 22:54:44 DEBUG (MainThread) [zigpy_cc.uart] Connecting on port /dev/ttyACM0 with boudrate 115200
2020-08-14 22:54:44 DEBUG (MainThread) [zigpy_cc.uart] Connection made
2020-08-14 22:54:45 DEBUG (MainThread) [zigpy_cc.api] --> SREQ SYS version tsn: None {}
2020-08-14 22:54:45 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x00!\x02#'
2020-08-14 22:54:45 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.SYS command_id=2 data=b'\x02\x00\x02\x06\x03\x90\x154\x01\x02\x00\x00\x00\x00' length=14 fcs=218>
2020-08-14 22:54:45 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP SYS version tsn: None {'transportrev': 2, 'product': 0, 'majorrel': 2, 'minorrel': 6, 'maintrel': 3, 'revision': 20190608}
2020-08-14 22:54:45 DEBUG (MainThread) [zigpy_cc.api] Version response: {'transportrev': 2, 'product': 0, 'majorrel': 2, 'minorrel': 6, 'maintrel': 3, 'revision': 20190608}
2020-08-14 22:54:45 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=persistent_notification, service=dismiss, service_data=notification_id=config_entry_discovery>
2020-08-14 22:54:45 INFO (MainThread) [homeassistant.setup] Setting up zha
2020-08-14 22:54:45 INFO (MainThread) [homeassistant.setup] Setup of domain zha took 0.0 seconds.
2020-08-14 22:54:45 DEBUG (MainThread) [zigpy.appdb] Loading application state from /config/zigbee.db
2020-08-14 22:54:45 INFO (MainThread) [zigpy_cc.zigbee.application] Starting zigpy-cc version: 0.4.4
2020-08-14 22:54:45 DEBUG (MainThread) [zigpy_cc.uart] Connecting on port /dev/ttyACM0 with boudrate 115200
2020-08-14 22:54:45 DEBUG (MainThread) [zigpy_cc.uart] Connection made
2020-08-14 22:54:46 DEBUG (MainThread) [zigpy_cc.api] --> SREQ SYS version tsn: None {}
2020-08-14 22:54:46 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x00!\x02#'
2020-08-14 22:54:46 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.SYS command_id=2 data=b'\x02\x00\x02\x06\x03\x90\x154\x01\x02\x00\x00\x00\x00' length=14 fcs=218>
2020-08-14 22:54:46 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP SYS version tsn: None {'transportrev': 2, 'product': 0, 'majorrel': 2, 'minorrel': 6, 'maintrel': 3, 'revision': 20190608}
2020-08-14 22:54:46 DEBUG (MainThread) [zigpy_cc.api] Version response: {'transportrev': 2, 'product': 0, 'majorrel': 2, 'minorrel': 6, 'maintrel': 3, 'revision': 20190608}
2020-08-14 22:54:46 INFO (MainThread) [zigpy_cc.zigbee.application] Detected znp version 'zStack12' ({'transportrev': 2, 'product': 0, 'majorrel': 2, 'minorrel': 6, 'maintrel': 3, 'revision': 20190608})
2020-08-14 22:54:46 INFO (MainThread) [zigpy_cc.zigbee.application] Forming network
2020-08-14 22:54:46 DEBUG (MainThread) [zigpy_cc.api] --> SREQ SYS osalNvRead tsn: None {'id': <NvItemsIds.ZNP_HAS_CONFIGURED_ZSTACK1: 3840>, 'offset': 0, 'len': 1, 'value': b'U'}
2020-08-14 22:54:46 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x03!\x08\x00\x0f\x00%'
2020-08-14 22:54:46 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.SYS command_id=8 data=b'\x00\x01U' length=3 fcs=62>
2020-08-14 22:54:46 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP SYS osalNvRead tsn: None {'status': 0, 'len': 1, 'value': b'U'}
[... cut ...]
2020-08-14 22:59:08 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF dataConfirm tsn: None {'status': 0, 'endpoint': 2, 'transid': 60}
2020-08-14 22:59:08 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: dataConfirm
2020-08-14 22:59:13 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xfaf7:2:0x0006]: command failed: off exception: 
2020-08-14 22:59:13 DEBUG (MainThread) [homeassistant.components.zha.entity] light.tyzb01_6sadkhcy_ts0002_e25c65fe_on_off_2: turned off: 
```

bajo, 2020-09-29:
I experienced this problem as well.

Ultraboss77, 2021-04-14:
Same issue here with CC2531 and Zemismart 2 gang w/o neutral