zigpy/zigpy-cc #75: Gledopto GL-C-007 2ID - issue with endpoint 15

jedi7, 2020-10-06:
**How to reproduce:**

* use TI CC2531 USB
* power up Gledopto GL-C-007 2ID (two devices in one)
* Add it to Home Assistant with the ZHA integration
* Device GL-C-007 with 2 entities should appear
* Try turn on/off the entities

**Expected results:**

Entity A (with endpoint 11, RGB/color) will on/off, etc
Entity B (with endpoint 15, CCT/white) will on/off, etc

**Actual results:**

Entity A (with endpoint 11, RGB/color) will on/off, etc
Entity B (with endpoint 15, CCT/white) does not respond (see logs)

**Relevant logs**
https://pastebin.com/5UgBGLn7

device description:
https://pastebin.com/LwyfS8mL

**Problem research**
When I tried the zigbee2mqtt, then it worked properly. And the difference is, the zigbee2mqtt uses always source endpoint 1.
So it looks like the firmware in Gledopto have issue when the source endpoint 15 is used.

When I patched it in code and forced source endpoint to 1, then I was able to change visually both entities (A,B). 
I.E. I was able to make send proper command, but response was not matched in code. And I'm not sure where to change it.

If you gave me some hints, I can try to change the code and made PR.

sanyatuning, 2020-10-06:
I think you need a specific zha device handler for this device.
I had the same problem with my xiaomi wall switches.
My solution
https://github.com/zigpy/zha-device-handlers/blame/dev/zhaquirks/xiaomi/__init__.py#L457

What was your patch?

jedi7, 2020-10-08:
@sanyatuning Many thanks! Based on the xiaomi quirk I created ugly quirk for gledopto.

https://pastebin.com/8q2NYfNW

I'm now able to on/off and change intensity. (no warm control for now)
I will continue on looking at it.

jedi7, 2020-10-08:
Can I close this issue as it is possible to solve it by quirks?

Hedda, 2021-01-12:
> Can I close this issue as it is possible to solve it by quirks?

Suggest that you close this if it has been solved via quirk in [zha-device-handlers](https://github.com/zigpy/)

closed 2021-01-12 by jedi7