zigpy/zigpy-cc #78: [BUG] Failed to connect to zzh! CC2652R

kmassada, 2021-04-07:
This is my first attempt at setting up ZHA.. for the CC2652R.. 

Steps to reproduce the behavior:

### Specs

```
Supervisor
Version
supervisor-2021.03.9

Core
Version
core-2021.3.4

OS Version:               Home Assistant OS 5.12
Home Assistant Core:      2021.3.4

3.13.3
NAME="Alpine Linux"
ID=alpine
VERSION_ID=3.13.3
PRETTY_NAME="Alpine Linux v3.13"
HOME_URL="https://alpinelinux.org/"
BUG_REPORT_URL="https://bugs.alpinelinux.org/"

Linux a0d7b954-ssh 5.9.16 #1 SMP PREEMPT Thu Feb 18 13:23:21 UTC 2021 aarch64 Linux
```

### Step 1 [Flash](https://electrolama.com/radio-docs/flash-cc-bsl/)
Installed pre-req for flashing

```
$ /usr/bin/python3 -m pip install --user pyserial intelhex
Collecting pyserial
Downloading https://files.pythonhosted.org/packages/07/bc/587a445451b253b285629263eb51c2d8e9bcea4fc97826266d186f96f558/pyserial-3.5-py2.py3-none-any.whl (90kB)
|████████████████████████████████| 92kB 2.5MB/s
Collecting intelhex
Downloading https://files.pythonhosted.org/packages/97/78/79461288da2b13ed0a13deb65c4ad1428acb674b95278fa9abf1cefe62a2/intelhex-2.3.0-py2.py3-none-any.whl (50kB)
|████████████████████████████████| 51kB 3.6MB/s
Installing collected packages: pyserial, intelhex
WARNING: The scripts pyserial-miniterm and pyserial-ports are installed in '/Users/usr/Library/Python/3.8/bin' which is not on PATH.
Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed intelhex-2.3.0 pyserial-3.5
WARNING: You are using pip version 19.2.3, however version 21.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

Confirm usb
```
usr-computer:cc2538-bsl-master usr$ ls /dev/tty.*
/dev/tty.Bluetooth-Incoming-Port	/dev/tty.usbserial-1410
```

Flashing successfully
```
usr-computer:Downloads usr$ python3 ./cc2538-bsl-master/cc2538-bsl.py -p /dev/tty.usbserial-1410 -evw CC2652R_coordinator_20210120.zip
Opening port /dev/tty.usbserial-1410, baud 500000
Reading data from CC2652R_coordinator_20210120.zip
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:22:81:A4:38
Performing mass erase
Erasing all main bank flash sectors
Erase done
Writing 171020 bytes starting at address 0x00000000
Write 148 bytes at 0x00029B780
Write done
Verifying by comparing CRC32 calculations.
Verified (match: 0x6a270c43)
**Expected behavior**
A clear and concise description of what you expected to happen.
```

### Step 2 [Pair ZHA](https://electrolama.com/radio-docs/zha-home-assistant/) 

USB  serial by id
```
~ ls /dev/serial/by-id/
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
```

Visible in dmesg
```
[ 7490.744693] ch341 1-1.1:1.0: ch341-uart converter detected
[ 7490.745393] usb 1-1.1: ch341-uart converter now attached to ttyUSB0
```

Also Visible in hardware options under supervisor..
```
- name: ttyUSB0
sysfs: >-
/sys/devices/platform/soc/ffe09000.usb/ff500000.usb/xhci-hcd.0.auto/usb1/1-1/1-1.1/1-1.1:1.0/ttyUSB0/tty/ttyUSB0
dev_path: /dev/ttyUSB0
subsystem: tty
by_id: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
attributes:
DEVLINKS: >-
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
/dev/serial/by-path/platform-xhci-hcd.0.auto-usb-0:1.1:1.0-port0
DEVNAME: /dev/ttyUSB0
DEVPATH: >-
/devices/platform/soc/ffe09000.usb/ff500000.usb/xhci-hcd.0.auto/usb1/1-1/1-1.1/1-1.1:1.0/ttyUSB0/tty/ttyUSB0
ID_BUS: usb
ID_MODEL: USB_Serial
ID_MODEL_ENC: USB\x20Serial
ID_MODEL_ID: '7523'
ID_PATH: 'platform-xhci-hcd.0.auto-usb-0:1.1:1.0'
ID_PATH_TAG: platform-xhci-hcd_0_auto-usb-0_1_1_1_0
ID_REVISION: '0264'
ID_SERIAL: 1a86_USB_Serial
ID_TYPE: generic
ID_USB_DRIVER: ch341
ID_USB_INTERFACES: ':ff0102:'
ID_USB_INTERFACE_NUM: '00'
ID_VENDOR: 1a86
ID_VENDOR_ENC: 1a86
ID_VENDOR_ID: 1a86
MAJOR: '188'
MINOR: '0'
SUBSYSTEM: tty
TAGS: ':systemd:'
USEC_INITIALIZED: '7491091504'
```

Attempting Connection results in generic "failed to connect" error. 

### Step 3 [ZHA Debug Logging](https://www.home-assistant.io/integrations/zha/#debug-logging)

```
2021-04-07 07:23:15 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event timer_out_of_sync[L]: seconds=1.8326559760000123>
2021-04-07 07:23:19 DEBUG (MainThread) [zigpy_znp.zigbee.application] Probing /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
2021-04-07 07:23:19 DEBUG (MainThread) [zigpy_znp.uart] Connecting to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 at 115200 baud
2021-04-07 07:23:19 DEBUG (MainThread) [zigpy_znp.uart] Opened /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 serial port
2021-04-07 07:23:19 DEBUG (MainThread) [zigpy_znp.uart] Toggling RTS/CTS to skip CC2652R bootloader
2021-04-07 07:23:19 DEBUG (MainThread) [zigpy_znp.uart] Connected to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 at 115200 baud
2021-04-07 07:23:19 DEBUG (MainThread) [zigpy_znp.api] Waiting 1s before sending anything
2021-04-07 07:23:20 DEBUG (MainThread) [zigpy_znp.api] Sending bootloader skip byte
2021-04-07 07:23:20 DEBUG (MainThread) [zigpy_znp.api] Waiting 1s or until a reset indication is received
2021-04-07 07:23:21 DEBUG (MainThread) [zigpy_znp.api] Testing connection to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
2021-04-07 07:23:21 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2021-04-07 07:23:24 DEBUG (MainThread) [zigpy_znp.api] Connection to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 failed, cleaning up
2021-04-07 07:23:24 DEBUG (MainThread) [zigpy_znp.uart] Closing serial port
2021-04-07 07:23:24 DEBUG (MainThread) [zigpy_znp.zigbee.application] Failed to probe ZNP radio with config {'path': '/dev/serial/by-id/usb-1a86_USB_Seri
al-if00-port0', 'flow_control': None, 'baudrate': 115200}
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/zigbee/application.py", line 159, in probe
    await znp.connect()
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/api.py", line 259, in connect
    self.capabilities = (await self.request(c.SYS.Ping.Req())).Capabilities
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/api.py", line 532, in request
    response = await response_future
asyncio.exceptions.CancelledError
 ```

Full gist: https://gist.github.com/kmassada/61d57ae4ca7116f3ebd852210798fca8

### Step 4 [Failing Test](https//gist.githubusercontent.com/omerk/0ee0e447a9e36786b4ff71d8f8126a23/raw/1474e89c284a5bd810e7bc0be05c1dab6eb4e71c/znp-uart-test.py)

```
➜  ~ sudo pip3 install pyserial
Collecting pyserial
  Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
     |████████████████████████████████| 90 kB 3.7 MB/s
Installing collected packages: pyserial
Successfully installed pyserial-3.5
➜  ~ python3 znp-uart-test.py
Got 0 bytes in response to PING command: b''
FAIL|Expected length mismatch
```

kmassada, 2021-04-07:
After treading the docs

> This script sends a ZNP command and checks that you get the expected response back. A PASS result means that you've sucessfully flashed your stick with the coordinator firmware. A FAIL result could either mean you have not flashed your stick properly, or you might not have specified the correct serial port (especially if you have multiple devices connected to you computer).

I decided to try again.. I did nothing new but try again several times

ran flash within 3 min of each other.. 

```
$ python3 ~/Downloads/cc2538-bsl-master/cc2538-bsl.py -p /dev/tty.usbserial-1420 -evw ~/Downloads/CC2652R_coordinator_20210120.zip
Opening port /dev/tty.usbserial-1420, baud 500000
Reading data from ~/Downloads/CC2652R_coordinator_20210120.zip
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:22:81:A4:38
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 171020 bytes starting at address 0x00000000
Write 148 bytes at 0x00029B780
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0x6a270c43)
$ python3 ~/Downloads/cc2538-bsl-master/znp-uart-test.py
Got 0 bytes in response to PING command: b''
```

3 min later

```
$ python3 ~/Downloads/cc2538-bsl-master/cc2538-bsl.py -p /dev/tty.usbserial-1420 -evw ~/Downloads/CC2652R_coordinator_20210120.hex
Opening port /dev/tty.usbserial-1420, baud 500000
Reading data from ~/Downloads/CC2652R_coordinator_20210120.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:22:81:A4:38
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F980
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0xc8b96313)
$ python3 ~/Downloads/cc2538-bsl-master/znp-uart-test.py
Got 7 bytes in response to PING command: b'\xfe\x02a\x01Y\x06='
PASS|OK
```

Also I've had to edit `znp-uart-test.py` to have the correct serial for mac `/dev/tty.usbserial-1420`...

closed 2021-04-07 by kmassada

cloudlakecho, 2021-07-06:
@kmassada Thanks for putting your issue and also sharing the solution by yourself. 

I have a question about your editing in the `znp-uart-test.py` to connect the specific device. I try to run the `test.py` in Zigpy-cc. How to choose a target device?  The reason is that I have two items in my `/dev` directory, `ttyACM0` and `ttyACM1`. And under the `/dev/seiral/by-id`, these are shown:
```
usb-Texas_Instruments_XDS110__03.00.00.11__Embed_with_CMSIS-DAP_L1100FCX-if00 -> ../../ttyACM0
usb-Texas_Instruments_XDS110__03.00.00.11__Embed_with_CMSIS-DAP_L1100FCX-if03 -> ../../ttyACM1
```

# ----- ----- ----- ----- ----- ----- ----- -----
This is my error message for your reference (when I execute `$ python test.py`): 

```
$ python test.py
asyncio DEBUG Using selector: EpollSelector
__main__ INFO STARTUP
zigpy_cc.zigbee.application INFO Starting zigpy-cc version: 0.5.2
Traceback (most recent call last):
  File "test.py", line 91, in <module>
    loop.run_until_complete(main())
  File "/home/ccho/anaconda3/envs/radec_finger_print_one/lib/python3.7/asyncio/base_events.py", line 587, in run_until_complete
    return future.result()
  File "test.py", line 85, in main
    await app.startup(auto_form=False)
  File "/home/ccho/project/zigpy-cc/zigpy_cc/zigbee/application.py", line 92, in startup
    self._api = await API.new(self, self._config[CONF_DEVICE])
  File "/home/ccho/project/zigpy-cc/zigpy_cc/api.py", line 94, in new
    await api.connect()
  File "/home/ccho/project/zigpy-cc/zigpy_cc/api.py", line 103, in connect
    self._uart = await uart.connect(self._config, self)
  File "/home/ccho/project/zigpy-cc/zigpy_cc/uart.py", line 185, in connect
    device = detect_port()
  File "/home/ccho/project/zigpy-cc/zigpy_cc/uart.py", line 170, in detect_port
    ", ".join(map(lambda d: str(d), devices))
serial.serialutil.SerialException: Unable to select TI CC device, multiple devices found: /dev/ttyACM1 - XDS110 (03.00.00.11) Embed with CMSIS-DAP, /dev/ttyACM0 - XDS110 (03.00.00.11) Embed with CMSIS-DAP
```

My runtime environment:
&nbsp;&nbsp; OS: Ubuntu 18.04
&nbsp;&nbsp; HW: CC2652R1 (Texas Instrument LaunchPad CC2642R1/CC2652R1 development kit)

Thanks.